OpenRTX/OpenRTX #1: DM-1701

v0l, 2020-06-09:
Hi,

What would be the process for testing this firmware on a new device, i dont really know where to start on this, i already tried to flash this on the device but it doesnt boot so i assume the offsets may be wrong or maybe even things are wired differently. 

Here is a picture of the internals:

[image]

n1zzo, 2020-06-09:
Hi @v0l, thank you for your interest,
right now the wrapping code in this project is only working for the md380,
however you could try building a firmware, using [the instructions in the readme](https://github.com/n1zzo/OpenDMR#installation), but after that you need to
sort out which kind of encryption is your radio using.
Then you could use [this tool](https://github.com/KG5RKI/CSFWTOOL) to wrap the code for your radio.

n1zzo, 2020-06-09:
I want to remind you that at the moment we are still at an early development stage, but if you're willing to build all our testsuites, that could be useful to see which hardware is configured differently on your radio. Still don't expect anything completely functional soon.

v0l, 2020-06-09:
Hey,

Thanks for the reply, i now have a starting point :D ill let you know if i make any progress.

silseva, 2020-06-09:
Hi,
here https://github.com/silseva/MD380-boot-tests there are also some boot tests we used during the very first development phase to check if everything worked correctly. 
You have to wrap the executable's binary also when using them.

v0l, 2020-06-12:
Hi,

This radio firmware is slightly different to what is described in md380tools, i took a dump of the USB wire traffic with USBPcap and recorded the firmware writing progress from the factory firmware upgrade tool, it goes like this:

```
Write Firmware:
    > 0x91 0x31 Prog mode?
    Erase: 0x4100000600
    Erase: 0x4100000700
    Erase: 0x4100000800
    Erase: 0x4100000900
    Erase: 0x4100000a00
    Erase: 0x4100000b00
    Erase: 0x4100000c00
    Erase: 0x4100000d00
    Erase: 0x4100000e00
    Erase: 0x4100000f00
    Set Address: 0x2100000600
    Write: 1024 bytes from wValue 0x02 to 0x41
    Set Address: 0x2100000700
    Write: 1024 bytes from wValue 0x02 to 0x41
    Set Address: 0x2100000800
    Write: 1024 bytes from wValue 0x02 to 0x41
    Set Address: 0x2100000900
    Write: 1024 bytes from wValue 0x02 to 0x41
    Set Address: 0x2100000a00
    Write: 1024 bytes from wValue 0x02 to 0x41
    Set Address: 0x2100000b00
    Write: 1024 bytes from wValue 0x02 to 0x36
    Erase: 0x4100c00008
    Erase: 0x4100000108
    Erase: 0x4100000208
    Erase: 0x4100000408
    Erase: 0x4100000608
    Erase: 0x4100000808
    Erase: 0x4100000a08
    Erase: 0x4100000c08
    Erase: 0x4100000e08
    Set Address: 0x2100c00008
    Write: 1024 bytes from wValue 0x02 to 0x11
    Set Address: 0x2100000108
    Write: 1024 bytes from wValue 0x02 to 0x41
    Set Address: 0x2100000208
    Write: 1024 bytes from wValue 0x02 to 0x81
    Set Address: 0x2100000408
    Write: 1024 bytes from wValue 0x02 to 0x81
    Set Address: 0x2100000608
    Write: 1024 bytes from wValue 0x02 to 0x81
    Set Address: 0x2100000808
    Write: 1024 bytes from wValue 0x02 to 0x81
    Set Address: 0x2100000a08
    Write: 1024 bytes from wValue 0x02 to 0x81
    Set Address: 0x2100000c08
    Write: 1024 bytes from wValue 0x02 to 0x81
    Set Address: 0x2100000e08
    Write: 1024 bytes from wValue 0x02 to 0x16
    Done
```

Once im able to write the stock image with md380-dfu i should be able to test the XOR to see if its still the same for these radios

v0l, 2020-07-28:
Well its been some time.. ive learned a lot haha, i managed to get `test_1` & `test_2` working on the DM1701 (today), so there is some more to learn... but for now that is my update. 

Also I created a tool (as practice for C++) to wrap/unwrap and flash firmware, you can see that project here: https://github.com/v0l/radio_tool

silseva, 2020-07-30:
Super!
Now are you trying to run also tests 3 and 4, I imagine. :)
Did you managed to find the schematic of the radio, for working on it?

PS: nice wrapping tool!

v0l, 2020-07-30:
Yes, i got 3 & 4 working now too, including OpenDMR, although the usb vcom doesnt appear to respond, im going to dismantle the radio again to try and work out how things are wired, it seems to be similar to the md380 anyway, status LED was the same at least... GPIOE PIN 0/1, maybe usb is on diff pins idk.. im starting to realize how steep this learning curve is haha..

```
[55737.767635] usb 1-1.4: new full-speed USB device number 30 using xhci_hcd
[55742.855829] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[55748.236025] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[55748.443965] usb 1-1.4: device not accepting address 30, error -62
[55748.444046] usb 1-1-port4: unable to enumerate USB device
```

v0l, 2020-07-31:
When i flashed OpenDMR on my dev board it seems to work though, strange..

```
[166878.233023] usb 1-1.2: new full-speed USB device number 37 using xhci_hcd
[166878.342838] usb 1-1.2: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
[166878.342839] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[166878.342840] usb 1-1.2: Product: STM32 Virtual ComPort in FS Mode
[166878.342840] usb 1-1.2: Manufacturer: STMicroelectronics
[166878.342841] usb 1-1.2: SerialNumber: 00000000050C
[166878.352673] cdc_acm 1-1.2:1.0: ttyACM1: USB ACM device
```

silseva, 2020-08-04:
Hmm... the pins should be the same, since the MCU is the same and there are no alternate functions for the USB ones. Looking back in our notes, I found we had the exact same problem you've now, meaning that we got the same error codes. Unfortunately, at the moment I can't remember what solved the issue, but you can try to make and flash an image from the master branch: it will not break anything because the makefile is currently configured to compile a testsuite for the internal RTC driver. The USB subsystem is initialised in any case, thus you can check if something change.

Let me know! :)

n1zzo, 2020-11-17:
Hi @v0l,
we did some major changes on this project since you last tried to run our code on your radio.
Now we have a solid modular structure based on uC/OS-III, a build system based on meson
(which in turn uses your own radio_tool script to flash ;) ), and we are working
towards having analog FM TX and RX at least on the MD-3[89]0 by end 2020.
If you want, you can follow the instructions in the `README.md`, try the different models to see
which best fits your radio, eventually we'll base your target on that model.

Have fun!

silseva, 2020-11-17:
@v0l the building target nearest to your radio is `mduv380g`, you can try directly that one. :)

73,
Silvano IU2KWO

v0l, 2020-11-17:
@n1zzo & @silseva cool, will give it a try!

Ive also created a release on github if you want to link to that for the CI builds https://github.com/v0l/radio_tool/releases/latest

n1zzo, 2020-11-27:
Thank you for the release, I'll try to make a meson command which automatically downloads the tool from your release.
We should have fixed the Virtual COM on the DM-1701, @v0l, can you confirm this so that we can close the issue?

v0l, 2020-11-30:
My DM1701 is not working right now, will try when i fix it :(

rigocalin, 2021-06-23:
> My DM1701 is not working right now, will try when i fix it :(

Hi @v0l as this thread is the close I could find to getting a custom firmware on DM1701, do you happen to have an update?

n1zzo, 2021-06-24:
@rigocalin are you a developer? You could try building OpenRTX for the closest platform and wrapping it with @v0l script, maybe something works out of the box. 
If you like the idea please hop on our Discord channel and we can talk about this there.

https://discord.com/invite/jZ9t8XTbmd

silseva, 2023-03-07:
I'm resurrecting this issue after Amo BD4VOW did some tests on the DM-1701. Now, thanks to his help, we have keyboard and screen set up and working, as long as part of the RF stage (if not all). Work in progress can be found here: https://github.com/OpenRTX/OpenRTX/tree/dm1701 and compilation target is `openrtx_dm1701`.

silseva, 2023-12-06:
Everything up to FM mode seems to work ok, M17 modulation is borked and needs investigation.

sp5wwp, 2025-02-04:
https://wiki.m17foundation.org/index.php?title=DM-1701
https://github.com/M17-Project/DM-1701_mod