OpenRTX/OpenRTX #2: FYI. Stack pointer add should probably be on a 4 byte boundary

rogerclarkmelbourne, 2020-06-26:
Guys

I noticed that you set the SP to 1 byte below the top of RAM.

https://github.com/n1zzo/OpenDMR/blob/master/linkerscripts/linker_script.ld#L15

FYI.
I found out that bootloader won't load the application if the SP is set to the top of RAM, so I understand you can't use the normal SP adddress

However, I'd advise that you change this to 4 bytes below the top of RAM, rather than 1 byte, because its a 32 bit processor and things just work better / faster if you align everything to 4 byte boundaries.

I know you loose 3 bytes of RAM if you do this, but I think its worth the sacrifice ;-)

silseva, 2020-06-27:
Hi Roger,
I know, it is no more than a leftover from the earlier testing stages. In any case thanks for the notice! :)

73,
Silvano IU2KWO.

closed 2020-06-27 by silseva

rogerclarkmelbourne, 2020-06-27:
No worries.  I noticed it was the same from the old tests...
It also too me a long time to work out why I could not load my test apps onto my MD380.
Until I disassembled the bootloader etc...