OpenRTX/OpenRTX #3: flash compile error

autodiag, 2020-11-13:
hello

i tyr final flash compile meson compile -C build_arm openrtx_md380_flash
get error ERROR: Can't invoke target `openrtx_md380_flash`: target not found but unknown what problem :(

n1zzo, 2020-11-13:
Hi autodiag,
thanks for the interest in our project,
the commands for compiling on the TYT MD380 are the following:
```
meson setup --cross-file cross_arm.txt build_md380
meson compile -C build_md380 flash_openrtx_md380
```
This should lead to a successful flash of the code to your radio,
please reply if you experience some issues during the process.

Nimayer, 2020-11-13:
@autodiag 
Before using the commands suggested by @n1zzo 
you need to compile and install `radio_tool` which is required to flash supported radios from Linux.

See the updated instruction in our [README](https://github.com/n1zzo/OpenRTX/blob/master/README.md) to do this.

silseva, 2021-03-19:
@autodiag did you managed to compile and flash OpenRTX on your radio?
Beware that, in the meantime, we made some changes to the overall structure, here you have the updated instruction for compiling and flashing: https://openrtx.org/#/compiling

n1zzo, 2021-03-23:
Closing this due to lack of response from issue author.

closed 2021-03-23 by n1zzo