OpenRTX/OpenRTX #4: MD-UV380/DM-1701: USB Virtual COM does not work

Nimayer, 2020-11-13:
USB Virtual COM (serial port over USB) does not work on Tytera MD-UV380 and Baofeng DM-1701

This is the Linux kernel dmesg shown after connecting DM-1701
```
[55737.767635] usb 1-1.4: new full-speed USB device number 30 using xhci_hcd
[55742.855829] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[55748.236025] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[55748.443965] usb 1-1.4: device not accepting address 30, error -62
[55748.444046] usb 1-1-port4: unable to enumerate USB device
```

silseva, 2020-11-15:
Regression also on MD-380:
`[10955.476495] usb 2-1.2: new full-speed USB device number 108 using ehci-pci
[10955.556538] usb 2-1.2: device descriptor read/64, error -32
[10955.744582] usb 2-1.2: device descriptor read/64, error -32
[10955.932729] usb 2-1.2: new full-speed USB device number 109 using ehci-pci
[10956.012768] usb 2-1.2: device descriptor read/64, error -32
[10956.200871] usb 2-1.2: device descriptor read/64, error -32
[10956.309202] usb 2-1-port2: attempt power cycle
[10956.913224] usb 2-1.2: new full-speed USB device number 110 using ehci-pci
[10957.329437] usb 2-1.2: device not accepting address 110, error -32
[10957.409476] usb 2-1.2: new full-speed USB device number 111 using ehci-pci
[10957.825610] usb 2-1.2: device not accepting address 111, error -32
[10957.825721] usb 2-1-port2: unable to enumerate USB device`

silseva, 2020-11-16:
Fixed regression for MD-380 in [42e4750](https://github.com/n1zzo/OpenRTX/commit/42e4750245b7b1d25654cc4714b8c330dd62f028).

silseva, 2020-11-17:
42e4750 also fixed the issue with TYT MD-UV380

closed 2020-12-19 by Nimayer