OpenRTX/OpenRTX #6: MD-UV380: Black screen after splash screen

Nimayer, 2020-11-17:
## Introduction
The current OpenRTX `main.c` should show a VFO interface.
This is done by employing several concurrent threads (tasks in uC/OS III naming) that manage various aspects of the radio:
* UI
* Radio baseband
* Platform state

## Problem
After introducing the threads in commit 67622d6cfa4645562a39c540514a2faca5448958 I observed a regression in the MD-UV380 radio.
When booting the radio I see the splash screen and then a black screen instead of the VFO screen.

## Other details
By experimenting with a demo I noticed two anomalies:
* The OS sleeps seemed to last much less than the specified time
* The device seems to hang after few seconds

silseva, 2020-11-17:
Bug reproduced also on MD-380. Code works correctly if compiled with arm-none-eabi-gcc 7.2.0, while fails with arm-none-eabi-gcc 10.2.0

silseva, 2020-11-17:
Fixed with b892e67

closed 2020-11-17 by silseva