OpenRTX/OpenRTX #11: MD390G: RX does not work

DC7IA, 2020-12-30:
I can transmit on the MD390G, but receiving does not work.

Neither the LED goes on, nor do I hear any audio.

DC7IA, 2021-01-02:
Since today, the microphone does not work anymore.

Yesterday I compiled and flashed the current code and it worked.
Today I pulled the changes and compiled and flashed again. Now I get a silent FM signal.

To make sure the microphone is not physically broken, I flashed back md380tools, which worked.

Looks like one of the commits today broke audio.

silseva, 2021-01-03:
@n1zzo tested the fix in his MD390G and now everything works.
If you still have troubles, let me know!

DC7IA, 2021-01-03:
RX still does not work.

DC7IA, 2021-01-03:
My fault, had repeater shift on 😂

closed 2021-01-03 by DC7IA