OpenRTX/OpenRTX #12: Cannot compile Linux target on aarch64

Nimayer, 2021-01-22:
Compiling the Linux target on an `aarch64` machine fails because the Linux uC/OS III port contains `x86-64` assembly code.

The compilation fails with the following error
```
pinephone:~/Clone-Room/OpenRTX$ meson compile -C build_linux openrtx_linux
Found runner: ['/usr/bin/ninja']
ninja: entering directory 'build_linux'
[1/2] Compiling C object openrtx_linux.p/platform_targets_linux_emulator_emulator.c.o
ninja: job failed: cc -Iopenrtx_linux.p -I. -I.. -I../openrtx/include -I../openrtx/include/calibration -I../platform/drivers/ADC -I../platform/drivers/NVM -I../platform/drivers/tones -I../openrtx/include/fonts/adafruit -I../platform/drivers/baseband -I../rtos/uC-OS3/Source -I../rtos/uC-OS3/Cfg -I../rtos/uC-CPU -I../rtos/uC-CPU/Cfg -I../rtos/uC-LIB -I../rtos/uC-LIB/Cfg -I../rtos/uC-OS3/Ports/POSIX -I../rtos/uC-CPU/Posix -I../platform/targets/linux -I../platform/targets/linux/emulator -I/usr/include/SDL2 -I/usr/include/directfb -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Wpedantic -g -pthread -D_REENTRANT -DDONT_USE_CMSIS_INIT -DSCREEN_WIDTH=160 -DSCREEN_HEIGHT=128 -DPIX_FMT_RGB565 -MD -MQ openrtx_linux.p/platform_targets_linux_emulator_emulator.c.o -MF openrtx_linux.p/platform_targets_linux_emulator_emulator.c.o.d -o openrtx_linux.p/platform_targets_linux_emulator_emulator.c.o -c ../platform/targets/linux/emulator/emulator.c
{standard input}: Assembler messages:
{standard input}:594: Error: unknown mnemonic `xorl' -- `xorl %ebp,%ebp'
{standard input}:595: Error: operand 1 must be an integer register -- `mov %RDX,%R9'
{standard input}:596: Error: unknown mnemonic `popq' -- `popq %rsi'
{standard input}:597: Error: operand 1 must be an integer register -- `mov %RSP,%RDX'
{standard input}:598: Error: operand 1 must be an integer or stack pointer register -- `and $~15,%RSP'
{standard input}:599: Error: unknown mnemonic `pushq' -- `pushq %rax'
{standard input}:600: Error: unknown mnemonic `pushq' -- `pushq %rsp'
{standard input}:601: Error: operand 1 must be an integer register -- `mov __libc_csu_fini@GOTPCREL(%rip),%R8'
{standard input}:602: Error: operand 1 must be an integer register -- `mov __libc_csu_init@GOTPCREL(%rip),%RCX'
{standard input}:603: Error: operand 1 must be an integer register -- `mov emulator_main@GOTPCREL(%rip),%RDI'
{standard input}:604: Error: unknown mnemonic `call' -- `call *__libc_start_main@GOTPCREL(%rip)'
{standard input}:605: Error: missing immediate expression at operand 1 -- `hlt'
ninja: subcommand failed
```

silseva, 2021-01-22:
The assembly block is here: https://github.com/OpenRTX/OpenRTX/blob/fa7c423edf81a5d2f4c716b8492900595a9d222e/platform/targets/linux/emulator/emulator.c#L136
It is needed to correctly initialise the libc before booting OpenRTX on x86_64 targets using an entrypoint whose name is not `main()`

Nimayer, 2021-04-02:
Current `master` branch b223235f2a5fd99f8e6f7498c90b4cdcfc29b05a compiles fine on Alpine Linux on `aarch64` (Tested on Pine64 PinePhone)

Just follow the updated instructions on the [OpenRTX website](https://openrtx.org/#/compiling) to install dependencies for Alpine Linux.

closed 2021-04-02 by Nimayer