OpenRTX/OpenRTX #13: Issues with MD-380

silseva, 2021-01-22:
On some versions of MD-380 targets we have:
- flipped screen
- RX not working
- TX working but with a ~20Hz noise superimposed to audio 

Affected devices having SW version S13.032 or above

sp5wwp, 2021-01-22:
TX works from 430.0000MHz to 438.8875MHz only. I didn't check lower frequencies.

tarxvftech, 2021-01-23:
TX works at 451.8M, but not RX.

tarxvftech, 2021-01-24:
For both devices:
MD380V (with GPS) with SN 160705088
MD380V (sans GPS) with SN 1507A15700

When flashed with 380 target, has a vertically mirrored display (but horizontally)
When flashed with 390 target, has a horizontally mirrored display (but vertically not mirrored)

tarxvftech, 2021-01-24:
for flipped screens: https://github.com/travisgoodspeed/md380tools/issues/178
and the further linked issues at the bottom too

silseva, 2021-01-24:
Thanks a lot!
I've encountered the issue on md380 tools and, also basing on this, we are planning to implement auto-detection of the display to have it correctly initialised on all the MDx devices.
We have found that original TYT firmware configures the display differently basing on a value stored in external flash's security register at address 0x301D: we are going to do the same on our firmware.

sp5wwp, 2021-01-24:
"TX working but with a ~20Hz noise superimposed to audio"

It was not the MD380's fault. You can remove that.

silseva, 2021-01-25:
Yep, thanks!
Recapping things, also basing on the new findings:
- flipped screen -> Confirmed, cause found (different display HW on each MD-380 HW revisions)
- RX not working -> Confirmed, cause found (will be solved when switching to RSSI-based squelch)

sp5wwp, 2021-01-26:
I can transmit at 430.0000, 431.0000 and 433.0000 MHz but I'm unable to do it at 432.0000.

That's part II of https://github.com/OpenRTX/OpenRTX/issues/13#issuecomment-765460596

sp5wwp, 2021-01-26:
It seems that transmitting at those frequencies results in a signal presence exactly 4.2 MHz below.

silseva, 2021-01-27:
Hi everyone!
I'm working on the screen issues you reported by making the display driver auto-detecting the hardware configuration by reading the value of security register 0x301D.
However, I need your help to check if everything is working properly, since I only have an "old vocoder" MD-380: please compile and flash the firmware from the MDx-display branch (https://github.com/OpenRTX/OpenRTX/tree/MDx-display) and let me know if the display orientation is still broken.
The instructions for compilation and flashing can be found here: https://github.com/OpenRTX/OpenRTX/wiki/How-to-compile

Thanks and 73,
Silvano IU2KWO.

tarxvftech, 2021-01-28:
@silseva : For my 3 single banders: your display branch works as expected on md380u, and on the MD380V and MD380VG is rotated 180 degrees (not mirrored, the display is upside down).

n1zzo, 2021-02-15:
@tarxvftech can you test the latest release v0.2 again on your MD380s?

tarxvftech, 2021-02-16:
(followed up on IRC, looks like a display issue on the VHF models, untested on the UHF model since it's display-less for the moment for other testing)

silseva, 2021-02-17:
@tarxvftech can you please check if commit 984f840 fixed the issue?

tarxvftech, 2021-02-18:
Will check in the next couple hours!

tarxvftech, 2021-02-18:
@silseva MD380VG and MD380V are both upside down on current master (984f8409f4462b381662b19783cc1cbdb741c2b6) (e.g. rotated 180*)

tarxvftech, 2021-02-18:
I'll poke around and see what I can do!

tarxvftech, 2021-03-18:
Said it over discord, but 0xE0 for posterity :)

silseva, 2021-03-19:
Updated to 0xE0 with commit 554c72c, thanks!
Now is everything fine.

closed 2021-03-19 by silseva