OpenRTX/OpenRTX #15: Battery level iusse

virtualrobotix, 2021-01-24:
During debug of my radio i saw that the level of battery go down when i push the ptt and going in transmission.
The battery in openrtx firmware is at middle level on original is on full level. 
When i push the level going down for all trasmitting period. 
If i continue to push and release push and release appear the battery icon and the radio shut down.
Thanks for your great work.
Roberto - IU2IOL -

DC7IA, 2021-01-24:
I've noticed that as well. It's because of voltage drop.

Just look at it when you're done transmitting. :D

Even my Yeasu FT-817ND does the exact same thing.

And I like an accurate reading. If voltage breaks in too much it could be a sign that the battery needs replacing.

n1zzo, 2021-02-15:
We could do something for the full charge indicator, the charging base stops somewhere around 7.1V to limit the battery wear,
I will adjust the battery meter to show 100% when battery is at 7.1V.

n1zzo, 2021-02-26:
We improved the battery level filtering and disabled the low battery check during transmission,
@virtualrobotix please test the latest master and if the issue is gone we can close this.

closed 2021-03-08 by n1zzo