OpenRTX/OpenRTX #19: RTOS / toolchain feasibility

nbgsmk, 2021-03-21:
Hello fellow hams and developers!
Thanks for your great work. This is an interesting and very poorly documented field. I understand the numerous obstac;les we have to pass, just to make a single radio boot up.
This is not an issue, but a query out of my own curosity as a developer.

Could you explain, for example: 
- OS: the background behind choosing a Miosix kernel for this development? Are we relying on some parts of the original radio firmware, and therefore need to build upon it, using the same OS? My question can also be formulated like this: without going into it's well known (dis)advantages right now, why not, for example, FreeRTOS?

- Toolchain: I'm not sure about each radio, but the majority of the interesting ones seem to be running STM32 these days. So, once more, I am curious why not using StmCubeIDE from ST?

- Hardware: I understand that manufacturers would like to prevent us from loading anything into their precious products (although 90% of the brands are not really manufactoring or developing anything :-) ). Therefore we have lack of documentation, obscure baseband chips, encrypted firmware images and other obstacles. So my question is: during the development phase, why not use JTAG interface to directly program the mcu? We could skip the whole wrapping, encrypting and downloading with yet-another-tool before running the latest code. When the work is done (if ever :-) ) the final fw image could be packed and released to the public, to be used with the radio's official upgrade tool.
Apart from easing the testing, this could also attract more developers who may be put off by the numerous steps involved and the steep learning curve for each. In my own personal case, I am quite interested to get involved, but not keen on spending a lot of time to set up the n-th toolchain for this-or-that, of the many things I am trying to tackle. On the other hand, for the stm32, I have the IDE, in-circuit debugger and free rtos knowledge, where I could start today. Ofcourse, they are all free of charge. Well, that's just my own personal oppinion.

By the way, I own an RT84 with specs quite similar to RT3s and md-uv380. The firmware for rt84 is identical to Baofeng 1701 (and another popular model, I can't remember right now). I prooved this by downloading from their manufacturer's support page and comparing the sha-512 checksums. RT84 uses an stm32 and, based on the checksum fact, it's a safe quess that both the mcu and the baseband hr_c6000 is also the same in those.

Thanks to anyone taking the time to comment. 

yt1zkb

silseva, 2021-03-21:
Hi!
Many thanks for your interest! It's always a pleasure to see that there is people interested in digging into our work! :)
So, here you are the answers you're looking for:

- OS: first of all, let me clarify that miosix kernel is not the same rtos of the original firmware. On the contrary, it is an open source project and you can find it here: www.miosix.org, https://github.com/fedetft/miosix-kernel.
However, we were using uC/OS-III until no more than one month ago, which is the new version of the rtos originally used by TYT (uC/OS-II). The reasons which led us to choose that one instead of another were not particularly significant, so I don't put them here.
More significant are the ones which led us to switch to miosix kernel: first of all, it is the only RTOS capable of using the newlib C standard library in a real multithreaded environment, whereas all the other RTOSes we evaluated lack this support or are doing ugly hacks to do so, which didn't suit our goal. Secondly, it is the only one which supports natively the POSIX pthread API, which allows us to write the code once and run it (almost) seamlessly on an x86/x64 Linux machine or on the radios, something really useful when developing the user interface.
Apart from POSIX support, the key point is that coupling the newlib shipped with arm-none-eabi-gcc with a multithreaded environment is not an obvious task: it requires support from the rtos to correctly manage the reentrancy data structures and the locking mechanisms required by malloc and other functions. FreeRTOS has a partial support for this, but to have everything set up correctly we would have needed to ship an ad-hoc newlib and, probably, also to modify some parts in the rtos.
Summarising: with miosix we have native support for multithreading either using the POSIX phtread API or the C++11 one and a truly thread safe C and C++ standard library also optimised for code size. To give you an example, on MD-380, only by switching from the old rtos with "plain" GCC to miosix we had the FLASH occupation decreasing from 13% to 9% (obviously using the same compiler optimisations).
The only "cost" to pay is that miosix comes with its own GCC, but the only way to obtain the features I've listed above is to patch the newlib and GCC sources. If you want to discover the details, you can look here: https://github.com/fedetft/miosix-kernel/tree/master/miosix/_tools/compiler/gcc-9.2.0-mp3.1/patches

- Two main reasons for this: the first one is that we have to develop and compile for x86/64, STM32F4 and NXP MK22FN512 (GD-77 and DM-1801), so we cannot simply use the StmCubeIDE, which is made only for ST MCUs (not considering that it will be quite difficult to integrate it with the miosix toolchain). The other reason for which we're not using the STM32 Cube APIs is that they're highly inefficient, both in terms of FLASH and RAM usage, and, also, because they are STM32-only. Since we're not supporting only STM32-based targets, the best way to be both efficient and platform-agnostic is to develop drivers which do exactly what we want and as we want, even more so if we consider that the development of these drivers is not a difficult task.

- In principle JTAG or, more precisely, SWD can be used for flashing/debugging. However, it has its troubles: the first one is that the radios lack an interfacing connector for it, they only have empty pads for pogo-pins. Thus, to have a permanent access to the debugging interface, one has to open its radio, solder jumper wires to bring the SWD lines outside and then close the cover back. The other issue is that, at least on MDx devices, the MCU pins where the SWD interface resides are also GPIOs effectively used to control some functionalities. As an example, on MD-380 the PA13 pin controls the microphone, while PA14 controls the 25/12.5kHz filter on the RX stage, and they are the SWD pins. 
To sum up, it is indeed possible to use SWD to flash the radio onboard MCU but, first of all, you have to solder jumper wires and, secondly, you cannot thoroughly test all the functionalities without loosing the debug interface.
In any case we try to keep the setup procedure as simple as possible, to allow people to play with the code and contribute. The only requirements are the miosix compiler, the meson build tool and radio_tool for flashing the radio without using the OEM's tool. Then, to edit the code, one can use its preferred editor and issue the commands for compile and flash directly from terminal.

Thanks for the info about the RT84! We can try porting OpenRTX to it, with your contributions if you're willing to.

73,
Silvano IU2KWO.

nbgsmk, 2021-03-21:
Wow! That is so much faster and more detailed than I expected! I really appreciate it. 
Admittedly, I am quite busy with the numerous interests (that I mandated upon myself), and it is questionable how usefull I can be to you, considering the new toolchains, setup, other stuff and the learning curve. 

I was not able to find a circuit diagram of rt84. If you have any knowledge in this direction, I would be glad to hear! Seeing no project dealing with rt84, and having no other good reason, I still haven't dissasemled mine. :)  So, no way to judge if swd pins are used or not..  Finding OpenRtx, I was willing to solder the connections, simply for the reason of avoiding numerous programming cable insertions. Like many others, mic/headset port on RT84 is the same as programming. This prevents me from using the radio after it is programmed, unless the cable is disconnected. I can imagine, say, 50 (dis)connections and develop-debug cycles in a day, but this would be really tiresome and ultimately damaging for it's connectors and pcb. If you add the regular day-speaker/evening-headset cycle, it becomes even worse.

On the other hand, if cheap enough and available locally, I may be willing to get any other model just for the programming stuff, and leave it permanently attached. I will try to get involved, but definitely please be kind to alert me additionally, if you plan to add RT84 or a similar radio, to your project.

I am particularly interested in RT84, obviously since I own it. But I was primarily looking for a dualband radio, for the dual-watch feature. Being on the same band or not, I find that monitoring two repeaters, two simplex, or whatever "at the same time" is usually 99% the same as owning two radios. Thats' why I am so excited by this project. Usually everyone focuses on monoband md-380, gd77 and that's all.  :-)

One thing that is bugging me is the stupid "zone" concept these days. I can still not accept that a zone is limited to the incredibly small 16, or barely insufficient 32 channels. Fortunately, RT84 has 64 channels in a zone and that is why I chose it. And still struggling with it! :) :) Ultimately, I would like to program all vhf/uhf, simplex, repeater and dmr channels into one single zone, even if that's 300 channels alltogether. To your knowledge, is there any reasonable technical limitation to the capacity of a zone? 
I have a simple method of remembering which freq is which memory channel, and I would navigate by typing eg. 123 (vhf simplex 23) or 223 (uhf simplex 32) on the keyboard. Unfortunately, just the analog part extends over 70 and adding dmr does not help my numbering scheme. I have to switch zones all the time, and it's a pain. If it was possible, just to have different zones in the A and B vfo, that would be a tremendous enhancement!

I have a few more ideas about the UI, but I don't want to overload you at once, without contributing something, too. That's the spirit, right? ;-)

Regards,
Zoran

n1zzo, 2021-03-22:
Hi Zoran,
setting up the toolchain is really something any linux user can do in half an hour, as @silseva said, you only need `meson`, `radio_tool` and the `MIOSIX` toolchain, please follow [the instructions on our website](https://openrtx.org/#/compiling). Regarding the learning curve you can reach out on Discord if you have doubts or cannot understand some architectural details of our project, so we can help you with that too.

Talking about the usage of programming cable vs SWD, try to think whether for you it is better to plug and unplug the kenwood connector, or have four wires, soldered on your PCB, which have to come out of the radio and be plugged into an SWD debugger.
In more than one year of OpenRTX development, we did thousands of flash cycles on our radios and not a single kenwood connector has failed until now. But if you prefer, get a second radio for development.

We don't possess the RT84, so the fastest way you can have that radio supported is either to donate one to the OpenRTX project, or to do the porting yourself. Obviously we will support you during the porting process.

Regarding the zone member limit, this is something related to the codeplug structure. In the radio codeplug, each zone is associated with an array of fixed size, which for example, in the MD380 is 16 elements long. Right now we chose to be compatible with the original codeplug format of each of our supported radios, so that OpenRTX users can switch back and forth between our firmware and the stock one without having to re-flash the codeplug. If scanning is your issue, a multi-zone scanning could be hypothesized as a workaround. But we have to implement scanning first.

Features proposals are as welcome as bug reports or direct contributions, feel free to open some issues, one for each feature, we'll discuss them case by case.

Check the manual that you have against the ones hosted [on our document repo](https://github.com/OpenRTX/OpenRTX-external-docs), if yours is not already present, reach out on Discord so that we can upload the document under the organization name.

nbgsmk, 2021-03-22:
Great info, and thanks a lot!
I am seriously considering it.
Regards!

n1zzo, 2021-03-23:
Thank you,

I am closing this issue.

closed 2021-03-23 by n1zzo