OpenRTX/OpenRTX #20: Selected menu item resets to "Zone" when gps is active.

CBJamo, 2021-03-24:
The selected menu item will jump back to the top, selecting "Zone". It seems to happen when the gps has been enabled. I thought it might happen when the gps was on but didn't have a fix, but I've observed the bug while the gps has a fix as well.

With the bug happening, I got the gps turned back off. Note that you can select other menu items if you time it right. With the gps off the bug immediately disappeared. When I re-enabled the gps, it was frozen with a previous fix. The radio otherwise seemed to be working correctly, though I only tested menus, not RX/TX.

Hardware: RT3s w/gps

Repro steps:
1. Power radio on
2. Enable GPS in a location where it can see SVs. It doesn't seem to occur when there are no sats.
3. Go to main menu, select any item other than zone.
4. Wait for the selected item to jump back to zone. Typically in a second or two.

silseva, 2021-03-26:
Fixed with commit 5219c00 and tested on my RT3s.

CBJamo, 2021-03-28:
Confirming fixed on my radio as well.

closed 2021-03-28 by CBJamo