OpenRTX/OpenRTX #21: Incorrect demodulation in UHF rx when bandwidth of 12.5 khz is used

andrealinux1, 2021-03-27:
I flashed the latest `master` on my `uv390` (which is basically identical to the `uv380`).

The rx and tx capabilities are mostly fine, but I noticed an incorrect demodulation, causing voice not to be intelligible, when the radio is used on the UHF band, in rx mode.

Thanks to the help of @Nimayer, we noticed that this problem arises only when a bandwidth of 12.5 kHz is used, therefore happening mostly when you use pre-saved channels.
If you start up your radio in VFO mode, with a bandwidth of 20/25 kHz, the demodulation is correct and you can hear voice fine.

Let me know if I can help you with testing/reproducing the bug, and thanks again for this fantastic project!

silseva, 2021-04-06:
Hi!
The bug should be fixed: if it is also for you, we can close the issue.

andrealinux1, 2021-04-08:
> Hi!
> The bug should be fixed: if it is also for you, we can close the issue.

Thanks for looking into this, I will compile the master and check the issues ASAP, and report back here!

andrealinux1, 2021-04-09:
Just tested, everything seems to work as expected! Mega Job!

silseva, 2021-04-09:
Good!
Thanks to you for the testing!

closed 2021-04-09 by silseva