OpenRTX/OpenRTX #23: I draw a MD-380 prototype diagram

songfei, 2021-04-26:
[image]

[openrtx.svg.zip](https://github.com/OpenRTX/OpenRTX/files/6374466/openrtx.svg.zip)

silseva, 2021-04-26:
Hi!
Many thanks! I think @cifred98 will appreciate this! :)

73,
Silvano IU2KWO.

cifred98, 2021-04-26:
Great work!
You may want to check out https://emulator.openrtx.org/ (It only works on Firefox/Chrome(ium) desktop and mobile Chrome)
My idea is to eventually emulate all the radios we support and having the SVGs is very convenient for us.
Thank you!

songfei, 2021-04-26:
I am very happy to hear this, and I will refine the buttons on the top and side.

Songfei BH1XWE

songfei, 2021-04-27:
version 2:

[image]
[openrtx.svg 2.zip](https://github.com/OpenRTX/OpenRTX/files/6382093/openrtx.svg.2.zip)

Nimayer, 2021-04-27:
Thank you @songfei, I like your drawings a lot!.
The only thing I would add is a tiny square near the channel/volume knobs to mimic the LED on the emulator.
Let me know what you think.

I would like to use your drawing also on the model pages on our website, like [this](https://openrtx.org/#/md380).
And they could also come handy when we will create a user manual.

Would you like to make a drawing of the MD9600 front panel too?

songfei, 2021-04-27:
I showed my painting to my designer friend, who is also ham, and he made some changes. 

I have thought about the position of the LED, it is not very beautiful in its original position, so I want to move it below.

[image]

songfei, 2021-04-27:
I haven't seen a real device of MD9600. May not be competent, I can draw DM1801 next, because I have this device.

Another flat style.

[image]
[openrtx.svg.zip](https://github.com/OpenRTX/OpenRTX/files/6384278/openrtx.svg.zip)