OpenRTX/OpenRTX #27: Can't compile

rd0fun, 2021-05-13:
Anyone know what this issue is?

%>meson setup --cross-file cross_arm.txt build_arm

ERROR: Malformed value in cross file variable cpp_link_args.

rd0fun, 2021-05-13:
OK I got past this... DUH... needs to be

~/.local/bin/meson setup --cross-file cross_arm.txt build_arm

But now I get another error....

`~/.local/bin/meson setup --cross-file cross_arm.txt build_arm
DEPRECATION: c_args in the [properties] section of the machine file is deprecated, use the [built-in options] section.
DEPRECATION: cpp_args in the [properties] section of the machine file is deprecated, use the [built-in options] section.
DEPRECATION: c_link_args in the [properties] section of the machine file is deprecated, use the [built-in options] section.
DEPRECATION: cpp_link_args in the [properties] section of the machine file is deprecated, use the [built-in options] section.
The Meson build system
Version: 0.58.0
Source dir: /home/rzach/OpenRTX
Build dir: /home/rzach/OpenRTX/build_arm
Build type: cross build
Project name: OpenRTX
Project version: 0.3.1

meson.build:4:0: ERROR: Unknown compiler(s): [['arm-miosix-eabi-gcc']]
The following exception(s) were encountered:
Running "arm-miosix-eabi-gcc --version" gave "[Errno 8] Exec format error: 'arm-miosix-eabi-gcc'"
`

silseva, 2021-05-13:
Hi @rd0fun,
is because you do not have the GCC toolchain for the miosix kernel installed on your system.
To install it, follow the installation instructions on our website: https://openrtx.org/#/compiling

73 de Silvano IU2KWO.

rd0fun, 2021-05-13:
that's the thing, I do have it installed. the binaries don't execute.

Nimayer, 2021-05-13:
@rd0fun, can you post the output of this command? `arm-miosix-eabi-gcc --version`

This way we know why your environment does not like the Miosix GCC.

Are you compiling on an architecture different than x86_64? That could be the cause since the Miosix tool chain currently supports only x64.

rd0fun, 2021-05-13:
Maybe that's the problem these are 64-bit executables?

silseva, 2021-05-13:
Sorry @Nimayer, I have to contradict you. The miosix wiki says: "The precompiled Miosix Toolchain is compiled for 32bit x86, to be compatible with as many Linux installations as possible", so the miosix GCC is x86 compatible.

@rd0fun can you tell me which version of Ubuntu you have? I have a x86 machine ad I can try replicate your setup.

Thanks!

rd0fun, 2021-05-13:
Ubuntu 18.04.5 LTS

silseva, 2021-05-13:
Ok, perfect. Another question, sorry: which is the output of the command `uname -a`?

rd0fun, 2021-05-13:
> Sorry @Nimayer, I have to contradict you. The miosix wiki says: "The precompiled Miosix Toolchain is compiled for 32bit x86, to be compatible with as many Linux installations as possible", so the [...]

Yeah that's what I read in the docs too.

rd0fun, 2021-05-13:
> Ok, perfect. Another question, sorry: which is the output of the command `uname -a`?

Linux ubuntu 4.15.0-143-generic #147-Ubuntu SMP Wed Apr 14 16:12:57 UTC 2021 i686 i686 i686 GNU/Linux

silseva, 2021-05-13:
Thanks! Tomorrow I'll try to install the toolchain on my x86 machine and I'll tell you the outcomes.

rd0fun, 2021-05-13:
> Thanks! Tomorrow I'll try to install the toolchain on my x86 machine and I'll tell you the outcomes.

thank you so much. I'm excited about this project! can't wait to play with it.

silseva, 2021-05-13:
In the meantime you can download and flash the binaries from the latest release, if you want! :wink: 
https://github.com/OpenRTX/OpenRTX/releases/tag/v0.3.1

silseva, 2021-05-14:
@rd0fun issue confirmed on my x86 machine.
It seems that, in spite of what is written in the miosix wiki, currently GCC is compiled for x64. I'll inform the maintainer and keep you updated.

rd0fun, 2021-05-14:
maybe a small note on the "compiling" page that says "64-bit only" would help others avoid my pitfall?

not sure why I didn't think of doing this earlier...but yes indeed, all 64-bit!

user@ubuntu:~$ file /opt/arm-miosix-eabi/bin/arm-miosix-eabi-gcc
/opt/arm-miosix-eabi/bin/arm-miosix-eabi-gcc: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 3.2.0, BuildID[sha1]=0276bc18600eaa62c10fbd154422104fb686c804, stripped

At least it wasn't me...  HA HA

silseva, 2021-05-16:
Hi!
Yesterday I talked with the maintainer of the miosix kernel: he told me that, when switched from GCC 4.7.3 to GCC 9.2.0, he dropped the support for 32 bit machines. This because, with all the linux versions switching to 64 bit, cross compiling for x86 became very difficult.
However, Phil DF5PMF provides nightly builds of the current master on his website: https://openrtx.schinken-radio.de/nightly/
If you want to compile OpenRTX on your machine, you have to compile the GCC compiler. This can be done by cloning the official miosix kernel repository and then by running the "download.sh" and "install-script.sh" shell scripts in the `miosix-kernel/miosix/_tools/compiler/gcc-9.2.0-mp3.1/` folder.
Before doing this, you have to uninstall the current version of the compiler you have on your machine. This can be done by running the "uninstall.sh" script located in `/opt/arm-miosix-eabi`

silseva, 2021-05-21:
@rd0fun have you tried to do something? :)

rd0fun, 2021-05-21:
> @rd0fun have you tried to do something? :)

Yes. I started from scratch with a clean Ubuntu 64-bit installation.

Still seeing failures in the instruction.

ubuntu@foo:~/OpenRTX/build_arm$ sudo apt-get install libstdc++6:i386 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libstdc++6:i386
E: Couldn't find any package by regex 'libstdc++6'

rd0fun, 2021-05-21:
I found it... apparently this is critical before adding this package!

sudo dpkg --add-architecture i386
sudo apt-get update

rd0fun, 2021-05-21:
Yeah. Compile success!

rd0fun, 2021-05-21:
Once ALL dependencies were satisfied, compile succeeded.

closed 2021-05-21 by rd0fun