OpenRTX/OpenRTX #28: Add Bluetooth BLE to UART module

songfei, 2021-05-18:
When I checked the PCB of MD-380, I found that the version without GPS has reserved a blank position. I hope to add a Bluetooth module here for UART communication between the radio and the mobile phone.

[image]

I found a Bluetooth module of similar size, made a converted PCB, and installed it on the radio.
Relevant information is here: https://www.ebyte.com/downpdf.aspx?id=1055

[image]

[image]

The program is now being debugged.

I want to hear your opinions, how about this function?

Nimayer, 2021-05-18:
Your idea and modification is truly impressive! 😮 
Adding Bluetooth BLE capability to an MD380 will open a lot of possibilities like remote control or updating the codeplug.

I have a question:
From what I know the original GPS module is only connected with UART RX signal, since the radio can read the GPS module serial port but cannot write to it.
Have you found a way to connect both TX and RX pins of the UART to your GPS module?

Also if you agree we can post a modification guide from you on [openrtx.org](openrtx.org) when the procedure is tested.

songfei, 2021-05-19:
I found a schematic diagram of MD-380. 
https://github.com/pchickey/md380-re/tree/master/datasheet

[image]

There are two wires connected to PD8 and PD9 of STM32F4. These two pins are RX and TX of UART3.

[image]

I think this can work.

silseva, 2021-05-19:
I imagined that, thanks anyway! :D
Let me go a bit deeper, just because we're curious: the original TYT firmware used PD9 as USART3_RX to receive data from the GPS module and PD8 as digital output to power it on and off. However, you're telling us that you've placed the BLE module on the same pad of an additional flash chip, thus effectively having direct access to both PD8 and PD9.
The question then is, again, just for curiosity, how do you manage the power control of the BLE module? The only possibility that comes to my mind is to have it permanently connected to the power supply rail and issue power on/off commands via the USART interface.

Thanks!
Silvano.

songfei, 2021-05-19:
Your guess is correct. Because of the circuit limitation, there are only 2 wires connected to the STM32, so when choosing the BLE module, I specifically found a module that can control the power supply through the serial port.

There is another problem. After the BLE module is connected to the mobile phone, the UART cannot use the AT command. At this time, you need to send the command to the mobile phone and let the mobile phone send the AT command to the BLE module.

I pasted a part of the BEL module documentation and translated it into English with Google.

> 4.2 Power mode
> The module supports two power modes: low-power mode and wake-up mode.
> 
> 4.2.1 Low power mode
> 
> The so-called low-power mode means that the BLE function continues to run after the module enters this mode, and the peripherals except the wake-up pin of the module are turned off. If you need lower power consumption, you can turn off broadcast and scan through AT commands, disconnect all connections, and set longer broadcast gaps, scan gaps, and connection gaps.
> Enter low power consumption:
**> 1. The AT command "AT+SLEEP" immediately enters the low power consumption mode;
> 2. AT command "AT+DISCSLEEP=1" set to enter low power consumption after disconnection;
> 3. AT command "AT+ONSLEEP=1" power on immediately enters low power consumption;**
> 4. Pass the rising edge of pin WKP and enter the low power consumption immediately after the high level is maintained for 200ms or more;
> After the module enters the low power consumption mode, it outputs "STA: sleep" through the serial port (LOGMSG does not turn off the output).
> Note: In the low-power mode, when the connection is not disconnected, such as ble receives air data, or when the connection status changes, the module temporarily wakes up and outputs the corresponding data, and immediately enters sleep after the data output is completed. At this time, low power consumption or wake-up will not output status data.
> 
> 4.2.2 Wake-up mode
> 
> The so-called wake-up mode means that the peripherals required by the module are in a normal working state in this mode. After the module wakes up, it outputs the status "STA: wakeup". Wake-up method:
> 1. Wake up immediately after the falling edge of the WKP pin, and the low level is maintained for more than 200ms;
**> 2. The serial port RX pin wakes up. The serial port rx falling edge, and the low level remains at 50us or above, wake up immediately.**

silseva, 2021-05-21:
It seems not to be a huge issue, anyway. If we make an OpenRTX app for remote control of the radio, then a proper mechanism for the shutdown of the module can be easily embedded into it.
Maybe when the app disconnects from the module there is a way to bring back the AT interface on the MCU side.

We keep waiting for updates, anyway: this idea is really cool! :D

73!
Silvano.

songfei, 2021-05-23:
I have finished the driver for the ble module. I would like to submit PR, this function is merged into the master branch or can you create a new branch?

YES，When the connection is disconnected, you can use the AT instruction on the MCU side.

I am not an embedded engineer, embedded software development and radio are my hobbies, my job is a mobile app development engineer, I would like to develop this APP.

silseva, 2021-05-23:
I think the best way is to create a separate branch, in this period we're making some structural changes in the codebase and having feature branches makes everything easier.

Of course you can develop the app! We have no reason to prevent you from doing so. :D

Nimayer, 2021-05-23:
Hi @songfei, if you want to develop an App to communicate with the radio it would be really great.

I kindly ask you to consider licensing the App source code with a Free and Open Source Software license like the GPLv3 we are using or another one of your choice.

Embracing the Free Software in your code makes sure that everyone in the community can benefit from and eventually contribute to it.

silseva, 2022-03-29:
Hi @songfei!
Did you have any chance to progress with the bluetooth interface?

songfei, 2022-04-06:
I am very sorry,  my job has been very busy recently. I have been following this openrtx project. When I have free time, I will continue to work.

silseva, 2022-04-06:
Don't worry, I asked just to know! :)
I perfectly understand you, by the way: I'm in a similar situation.

Thanks!

closed 2026-03-10 by github-actions[bot]