OpenRTX/OpenRTX #29: Need for NF filtering?

criede, 2021-05-30:
Hello,
on a MMDVM board, there are a two stage opamp filter and two level adjust potentiometers on the receive path in front of the A/D-Converter of the microcontroller. As far as I understand, in your suggested MD380-Mod, you are connecting the discrimiminator more or less directly to the A/D-Pin of the STM. Why is in your case no need for filtering and level adjustment?
Thanks
Christian DB9CR

silseva, 2021-05-30:
Hello Christian!
I have to anticipate that I do not know how the MMDVM is structured, but I'll take a look a the schematic.
This said, we decided to connect the STM32 ADC almost directly after the discriminator output because, to correctly decode M17, we need an audio path ensuring flat frequency response from almost DC to 2.4kHz.
The first way we investigated was the one of sampling with two ADCs the QT_DQT_IN and 2T/5T signals and then combining the samples to reconstruct the full bandwidth of the signal: this approach, however, is a bit difficult as we also have to take into account the presence of a phase difference between the two signals caused by the various filtering stages. The QT_DQT_IN signal, for instance, goes at maximum up to 300Hz or so, as in the original design it is dedicated only for decoding of an incoming CTC/DCS signal. The other one ranges from ~300Hz to ~4kHz, if I remember correctly.
When looking for an alternative approach and after a bit of testing from Wojciech SP5WWP, it emerged that the signal on pin 1 of U102 has all the required characteristics for a good decoding of the incoming M17 baseband signal, thus we settled on the solution of directly connecting that pin to the MCU. The test performed consisted in connecting and USB soundcard to U102 and decode in real-time the incoming samples using m17-cxx-demod, and it worked (https://www.youtube.com/watch?v=-dnLuYshmjY).
So, given all this, we have elements to be reasonably sure that connecting the STM32 ADC to the pin 1 of U102 allows us to sample the incoming signal with the correct levels and without distortion. Another point is that this patch has the advantages of requiring a minimal intervention on the hardware and of not having a significant impact (to not say none) on the operation with the original firmware.

I hope I have answered to your doubts: if not, feel free to ask! :)

73,
Silvano IU2KWO.

criede, 2021-06-07:
Thanks for your answer - looking forward to see M17 in action.

silseva, 2021-06-11:
We're working on it, stay tuned! ;-)

73,
Silvano IU2KWO.

closed 2021-06-11 by silseva