OpenRTX/OpenRTX #32: encryption

jimgillo, 2021-06-29:
Hello,

Wy don't you modify the OpenRTX code, to allow to add new encryption algorithm if we want to use better crypto ?

silseva, 2021-06-29:
Hi!
To which encryption are you referring? The DMR one?

jimgillo, 2021-06-30:
The DMR crypto in TYT MD-380/390 is very bad, even the 128 bit crypto.

Travisgoospeed shows that the crypto uses the AES-128 cipher to create a 49bit stream (always the same with the same key) which is XOR'ed with ALL 49bit voice stream until the key is changed.
Such cipher can be cracked with analyze of null frames, there is no security. It's like an old vigenere cipher.

Tavisgoodspeed write in the https://github.com/travisgoodspeed/md380tools/blob/master/applet/src/aes.c

/* The key has more than its least-significant byte set, so we'll

use the original Tytera algorithm. At some point, it might make

sense to replace this with proper crypto, rather than XOR.

*/
I asked him if he could create better crypto or to add some hook in the source code but he has no time for it.

The hook in the aes.c code is not enough, to add better crypto we must access to the hook where the code XOR the AES stream named res with the voice stream.
There is no hook in the code to access this part.

printf("\nOUT :");

res=aes_cipher(pkt);

printhex(res,16); //Print the keystream it produces. (First 49 bits are XOR'ed with the audio.)

printf("\n");

return res;

the 49bit key stream is res, but we don't have access to the part of the firmware where this stream is XOR-ed with the voice frame.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

DC7IA, 2021-07-19:
Yeah, that does not matter at all for amateur radio.

silseva, 2021-08-10:
Aside from the fact that in some countries, like Italy, encrypted ham radio communication is forbidden, once support for DMR will be up and running you will be able to implement whatever encryption you deem adequate! :)
Our implementation does not rely on any part of the original TYT firmware, so you'll have access to every data stream to and from the DMR baseband chip, being it either voice or data (SMS).

n1zzo, 2021-09-17:
Closing as DMR support is not yet underway.

closed 2021-09-17 by n1zzo