OpenRTX/OpenRTX #33: DMR

NeleBrt, 2021-06-30:
Will you add dmr mode to OpenRTX, and when?

closed 2021-07-01 by NeleBrt

n1zzo, 2021-09-17:
Hi @NeleBrt,
DMR support requires a working AMBE+2 codec, and that is presently patented, therefore we cannot implement the DMR functionality without a commercial agreement with DVSI, that holds the commercial rights of the patent.
Until the legal issues relative to the AMBE codec will be solved we will focus on free protocols such as [M17](https://m17project.org/).

majkrzak, 2022-05-08:
No properitary code have to be used, as there seems to be some opensource implementations which can be utilised here. In the worst case, some effort to implement it by ourselves can be taken.