OpenRTX/OpenRTX #42: Decode CTCSS tone not working on md-uv3x0

Nimayer, 2021-10-13:
Tested on: Retevis RT3s

The CTCSS tone is set to D or E+D from the macro menu.
When transmitting with a second radio with the corresponding TX CTCSS tone (71.9 in my case) enabled, 
no sound is heard on the receiving radio.

Tyler-2, 2022-01-16:
Confirmed bug on TYT MD UV-390 with nightly v0.3.2-82-g2bb05a2f also.

silseva, 2022-04-15:
We'll fix it as soon as possible, thanks for reporting! :)

silseva, 2022-08-02:
Finally solved.

closed 2022-08-02 by silseva

Tyler-2, 2022-09-08:
Are the TYT MD-UV390 builds not getting updated?

I just tried it on my lunch break with `v0.3.2-82-g2bb05a2f` freshly downloaded off Phil's nightly page and it seems like it's still not working.... and it turns out that's the same build as my original complaint!

silseva, 2022-09-09:
I think is a problem with Phil's nightly builds system, I noticed the same.
We now have our nightly build server too, the firmware images can be found here: https://files.openrtx.org/nightly/.
Alternatively, you can flash the binaries from the latest release (v0.3.4), which can be found on the repository main page.