OpenRTX/OpenRTX #49: trouble compiling

DC7IA, 2021-11-12:
Hi,

last time I compiled OpenRTX (10 months ago), I did it like this:

```
meson setup --cross-file cross_arm.txt build_arm
ninja -C build_arm openrtx_md390_bin -j2
radio_tool --wrap -o build_arm/openrtx_md390_wrap -r MD380 -s 0x0800C000:build_arm/openrtx_md390_bin
radio_tool -d 0 -f -i build_arm/openrtx_md390_wrap
```

However, now 

```
meson setup --cross-file cross_arm.txt build_arm
```
results in

```
ERROR: Malformed value in cross file variable cpp_link_args.
```
and 

```
meson compile -C build_arm openrtx_MD390
```

results in

```
usage: meson [-h] {setup,configure,dist,install,introspect,init,test,wrap,subprojects,help,rewrite} ...
meson: error: unrecognized arguments: -C build_arm openrtx_MD390
```

Have there been any changes in the last 10 months that mean I need to do something differently?

Tried on Ubuntu 20.04 LTS running 

```
% meson --version
0.53.2
% ninja --version
1.10.0
```

I'm not a dev, so I don't really know what this is trying to tell me. :/

silseva, 2021-11-12:
Hi!
Does meson tell you if the error `ERROR: Malformed value in cross file variable cpp_link_args.` is associated to a line in a file?
Alernatively, the folder `build_arm` should contain a file named "meson.log" or something like this: if you find it, please post it here.

About the other point: starting from the release v0.2, we merged the MD380 and MD390 targets into a single `MD3x0` one. The new compiling command, thus, became: `meson compile -C build_arm openrtx_MD3x0`

DC7IA, 2021-11-12:
>  Alernatively, the folder `build_arm` should contain a file named "meson.log" or something like this: if you find it, please post it here.

```
% ls -a *
meson-info:
.  ..

meson-logs:
.  ..

meson-private:
.  ..
% ls -a
.  ..  meson-info  meson-logs  meson-private
```

That folder is empty, other than the three subfolders.

> Does meson tell you if the error `ERROR: Malformed value in cross file variable cpp_link_args.` is associated to a line in a file?

That is the complete error message. :thinking:

silseva, 2021-11-12:
Ok, no problem!

silseva, 2021-11-12:
I just found out that another user had the same issue months ago (and I forgot that): https://github.com/OpenRTX/OpenRTX/issues/27
Basically the point is that we switched from FreeRTOS to another RTOS and this one comes with a patched version of the ARM GCC compiler, which has to be installed on your machine to compile everything. The issue I linked above details it a bit more.

If you don't want to install the toolchain and you are interested only in using OpenRTX (and not modifying/experimenting with the sources), you can flash the nightly builds compiled from the master branch. You can find them here: https://openrtx.schinken-radio.de/nightly/

DC7IA, 2021-11-12:
> I just found out that another user had the same issue months ago (and I forgot that): #27

```
wget https://miosix.org/toolchain/MiosixToolchainInstaller.run
sh MiosixToolchainInstaller.run
```

Software shouldn't be installed from places other than the official repos.

Debian officially says this could break Debian: https://wiki.debian.org/DontBreakDebian

Same applies for Ubuntu as it is based on Debian.

Is there another way?

I don't want to run a script that places stuff in `/usr/bin` and `/opt`.

Can I place the binaries somewhere in `/home/` and tell meson to use the binaries there?

Otherwise I will have to use a virtual machine to keep my production system sane and don't break things.

silseva, 2021-11-12:
> Can I place the binaries somewhere in /home/ and tell meson to use the binaries there?

Yes, is possible.

Quoting from the miosix wiki (https://miosix.org/wiki/index.php?title=Linux_Quick_Start):

> If you do not trust the installer and want to verify its content, or you want to install it locally, it is possible to extract the content of the installer with the following command.
>`sh MiosixToolchainInstaller.run --noexec --target arm-miosix-eabi`

You can extract the binaries in your home directory (or in any other place you prefer) and then edit the first lines of the `cross_arm.txt` file so that the contain the path to the binaries. This should work.

In the meantime we are alread working to provide an alternative way to compile OpenRTX for those ones that prefer not to install the miosix toolchain by making meson downloading a local copy of the executables and run those. The executables, in turn, are compiled via CI on our for of the miosix kernel repository.

DC7IA, 2021-11-12:
> In the meantime we are alread working to provide an alternative way to compile OpenRTX for those ones that prefer not to install the miosix toolchain by making meson downloading a local copy of the executables and run those.

Looking forward to that!

Only reason I'm compiling myself is so that I can look at the M17 branches.

If you'd provide releases for that as well I'd use those instead.

Thanks for your help, now I'm able to compile it all. :)

DC7IA, 2021-11-12:
Do you maybe know where I can find a repo for the Miosix Toolchain?

That way I can build a little script that pulls updates and `make`s the toolchain before compiling from the latest sources, so everything will always be up to date.

DC7IA, 2021-11-12:
```
~/git/radio_tool/build/radio_tool --wrap -o build_arm/openrtx_md3x0_wrap -r MD380 -s 0x0800C000:build_arm/openrtx_md3x0_bin

results in:

```
Error: Cant open file for segment
```

any advice?

silseva, 2021-11-12:
The compilation scripts for the Miosix toolchain are located in the same repository containing the Miosix's sources. Here it is the path for the last compiler version: https://github.com/fedetft/miosix-kernel/tree/master/miosix/_tools/compiler/gcc-9.2.0-mp3.1

As you will see, the toolchain is updated once in a blue moon so, probably, we just need to store the binaries somewhere and make meson download them if it does not detect an already installed toolchain in the system.

About M17 support: the code for transmission is already merged in the `master` branch and, consequently, the nightly builds are M17-capable. You will not find RX/decoding because we're implementing it in these weeks.

> any advice?

I have to look at the problem once I get back home, will do this evening!

DC7IA, 2021-11-12:
> The compilation scripts for the Miosix toolchain are located in the same repository containing the Miosix's sources. Here it is the path for the last compiler version: https://github.com/fedetft/miosix-kernel/tree/master/miosix/_tools/compiler/gcc-9.2.0-mp3.1

Thank you! I've searched so long, but couldn't find it!

> About M17 support: the code for transmission is already merged in the `master` branch and, consequently, the nightly builds are M17-capable. You will not find RX/decoding because we're implementing it in these weeks.

Does the MD390G need to have modifications? The documentation does not mention it at all. :thinking: 

> I have to look at the problem once I get back home, will do this evening!

Thanks :)

silseva, 2021-11-12:
> Does the MD390G need to have modifications? The documentation does not mention it at all. 🤔

Yes, the same modifications listed for the MD380: internally they have exactly the same hardware.
We have to make the website more clear, about this.

DC7IA, 2021-11-12:
> Yes, the same modifications listed for the MD380: internally they have exactly the same hardware.

Can I do that without it no longer being waterproof? :D

silseva, 2021-11-12:
> ```
> ~/git/radio_tool/build/radio_tool --wrap -o build_arm/openrtx_md3x0_wrap -r MD380 -s 0x0800C000:build_arm/openrtx_md3x0_bin
> ```
> 
> results in:
> 
> ```
> Error: Cant open file for segment
> ```
> 
> any advice?

Tested on my machine, and it works. But i think that the problem lies in the compile command you issue, which made you skip the stage generating a binary file.
Summarizing all the commands, from the beginning:

```
meson setup --cross-file cross_arm.txt build_arm
ninja -C build_arm openrtx_md3x0_bin -j2
radio_tool --wrap -o build_arm/openrtx_md3x0_wrap -r MD380 -s 0x0800C000:build_arm/openrtx_md3x0_bin
radio_tool -d 0 -f -i build_arm/openrtx_md3x0_wrap
```

Generally speaking:
- `meson -C build_arm openrtx_md3x0` -> compiles all the sources and generates an ELF image
- `meson -C build_arm openrtx_md3x0_bin` -> compiles all the sources and generates an ELF and a binary image
- `meson -C build_arm openrtx_md3x0_wrap` -> as above and generates ELF + binary + wrapped binary for the bootloader

PS: your current version of meson should support all the native commands also written in our website, so I think you do not need to directly call ninja anymore! :)

silseva, 2021-11-12:
> > Yes, the same modifications listed for the MD380: internally they have exactly the same hardware.
> 
> Can I do that without it no longer being waterproof? :D

Well, the only way to be completely sure that the radio will remain waterproof is not opening it...

Apart from that, I think that if you are careful during disassembly and reassembly, your radio should remain reasonably waterproof.

closed 2021-11-12 by DC7IA