OpenRTX/OpenRTX #50: Display brightness setting

sp5wwp, 2021-11-12:
Since settings are not volatile anymore, you should prevent users from turning off the display's backlight (PWM=0).

nolith, 2021-11-19:
On the MD380, if it happens you can always do M+7 to bump up the brightness. (but you have to remember)

What where you thinking about? Preventing backlight to go to 0, or always bumping it to the min level at boot time if below that?

silseva, 2021-11-20:
For now let's implement the second one, that is bumping the brightness to a minimum level if a value of zero was saved. In this way we leave the possibility to set it to a very low level during operation, useful for those with delicate eyes.

silseva, 2021-11-20:
Feature/fix implemented by @nolith, we can close the issue.
Thanks! :D

closed 2021-11-20 by silseva