OpenRTX/OpenRTX #52: MD380: redio does not transmit with the external PTT signal

nolith, 2021-11-19:
While recording the videos on https://github.com/OpenRTX/OpenRTX/issues/40#issuecomment-974110862 I realized than once I flashed the radio with OpenRTX it was no longer transmitting with the external PPT signal. As can be see in the second video, I had to pick the radio in my hand and press the PTT on the radio body.

nolith, 2021-11-19:
I did a second test with another PTT button.

It works with the stock fw
[image]

But not with OpenRTX

[image]

nolith, 2021-11-20:
As @silseva pointed out on Discord, there is no pull-up resistor on the RT3 and the internal one has to be enabled on the gpio port like in an MD-UV380

silseva, 2021-11-20:
Fixed with commit dccefd0

closed 2021-11-20 by silseva