OpenRTX/OpenRTX #66: TYT MD-UV390 can only monitor single channel?

Tyler-2, 2022-01-16:
I didn't see this in the capability notes, and I didn't see that there was a way to restore this functionality in the user docs... But is it accurate that OpenRDX right now doesn't work to monitor multiple channels? I'm sorry i don't know the name for this... but in the original firmware you had two "slots" that could independently be channels/frequencies, and you could switch between them and tune them individually. You could for instance have two repeaters in those slots, so that if there were any transmissions from either repeater you'd hear it. It's what the up/down arrows are for on the main screen, switching between those two channels.

silseva, 2022-01-17:
Hi!
You're right, we did not implement the dual watch functionality yet. We'll add it in the next updates, we're planning to restructure and update a significant part of the codebase.

closed 2022-03-29 by silseva

Tyler-2, 2022-03-29:
Was this closed because it was added in recent updates? If so, exciting!

silseva, 2022-04-06:
Not yet, I'm sorry. In any case is recorded in our internal todo list as a feature to be added after the codebase reorganisation, which is going to start soon! ;)