OpenRTX/OpenRTX #71: RT3S can't update codeplug

kmihaylov, 2022-04-11:
Just flashed the latest fw [5ffd6e5](https://github.com/OpenRTX/OpenRTX/commit/5ffd6e5c437e2c18aabfbdce996f4c566685b93e).

Unfortunately I can't update my codeplug. I tried with the original Retevis software and also with qdmr. Qdmr says recognizes the radio and also when I click "Verify" it returns success.

silseva, 2022-04-15:
Yes, this issue is due to the fact that, for various reasons, we have not implemented CPS transfer yet.
However we're actively working on it and soon there will be an ad-hoc OpenRTX CPS structure with data writable using Qdmr and other tools, we already contacted the maintainers.

kmihaylov, 2022-04-15:
I have some STM experience. Would you please give me some directions how is the codeplug read by the radio? I did quick view on the code and I can't understand what happens where yet.

Offtopic, is DMR mode supported?

silseva, 2022-05-04:
The driver which reads codeplug data from the external flash is located inside platform/drivers/NVM folder. It will became obsolete soon, though, because we're moving towards having LittleFS on the external flash and save everything as files.

About DMR: is not supported yet because the AMBE+2 codec is still protected by a patent and including the binary image from the original firmware will be a patent infringment.

silseva, 2022-09-06:
Closing for inactivity, feel free to reopen.

closed 2022-09-06 by silseva