OpenRTX/OpenRTX #72: RT3S incorrect voltage

kmihaylov, 2022-04-11:
My battery indicator is (almost) fully green. Settings->Info gives 85%. Voltage is 6.9 to 6.10 (something wrong here...?).
However I checked the battery terminals w/ the multimeter and the voltage is 7.16V. How does the firmware checks the voltage?

silseva, 2022-04-15:
Hi!
Voltage is measured using an ADC channel connected to a voltage divider directly connected to the battery.
I did a quick test now on my MD380: the battery is at 7.44V and the radio reports 7.6V, which is fine. I cannot exclude that there are bugs for the other radios, however. Which model do you have?

silseva, 2022-04-15:
Tested also on my RT3S compiling from the latest commit on master branch: battery voltage is 7.43V, radio reports 7.3V and battery level 25%.

Here you can find our nightlies too: https://files.openrtx.org/nightly/

kmihaylov, 2022-04-15:
The battery indicator was fully black. Then it became fully green. This happened only once. I'll watch it.

I'll close the issue in day or two. Last time I checked the voltage was OK.

silseva, 2022-05-18:
Hi @kmihaylov!
Can we considered the issue solved or it needs further investigation?

sp5wwp, 2022-06-11:
I've got a similar (or maybe the same) problem with MD380. Seems like the code treats 7.10 as a higher value than 7.9.
[image]

sp5wwp, 2022-06-11:
Most likely, the bug was in the integer part calculation. I have already opened a PR with a fix.

closed 2022-07-10 by silseva