OpenRTX/OpenRTX #81: GD-77/DM-1801: GLIBC version of scripts/bin2sgl.Linux.bin

jg1uaa, 2022-06-12:
Hi, I am trying to build OpenRTX for DM-1801 on Debian11.3/amd64 virtual machine.
But it aborts with following error.

```
uaa@debian-vm:~/OpenRTX$ meson compile -C build_arm openrtx_gd77_flash
ninja: Entering directory `/home/uaa/OpenRTX/build_arm'
[68/71] Linking target openrtx_gd77
Memory region         Used Size  Region Size  %age Used
           flash:      141648 B       494 KB     28.00%
             ram:        6128 B       128 KB      4.68%
[70/71] Generating openrtx_gd77_wrap with a custom command
FAILED: openrtx_gd77_wrap.sgl
/home/uaa/OpenRTX/scripts/bin2sgl.Linux -f openrtx_gd77_bin && mv openrtx_gd77_bin.sgl openrtx_gd77_wrap.sgl
/home/uaa/OpenRTX/scripts/bin2sgl.Linux.bin: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /home/uaa/OpenRTX/scripts/bin2sgl.Linux.bin)
/home/uaa/OpenRTX/scripts/bin2sgl.Linux.bin: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /home/uaa/OpenRTX/scripts/bin2sgl.Linux.bin)
ninja: build stopped: subcommand failed.
uaa@debian-vm:~/OpenRTX$ 

```

I think it can be solved with rebuilding this binary. Is there any source code for this?
And, target dm1801_flash is not accepted.

```
uaa@debian-vm:~/OpenRTX$ meson compile -C build_arm openrtx_dm1801_flash
ninja: Entering directory `/home/uaa/OpenRTX/build_arm'
[2/3] Generating openrtx_dm1801_wrap with a custom command
Adding segment 0x0800c000 from file openrtx_dm1801_bin
Done!
[3/3] Generating openrtx_dm1801_flash with a custom command
FAILED: openrtx_dm1801_flash
/home/uaa/OpenRTX/subprojects/radio_tool/./radio_tool -d 0 -f -i openrtx_dm1801_wrap
Error: Radio not supported
ninja: build stopped: subcommand failed.
uaa@debian-vm:~/OpenRTX$ 
```

silseva, 2022-06-12:
Hi!
For the first point: the linux binary for bin2sgl in the repo was compiled with a too recent version of the libc (our fault, we all use arch linux and it has the latest libraries). In any case, as you said, you just have to recompile it. The source code is available on the OpenGD77 repository at this URL: https://github.com/open-ham/OpenGD77/blob/main/tools/bin2sgl/bin2sgl/bin2sgl.c

For the second point: our fault again, the meson.build is incomplete. We'll fix it as soon as possible, in the meantime you can use the following workaround:

- issue the command `meson compile -C build_arm openrtx_dm1801_bin` to get a compiled, unencrypted, binary image;
- issue separately the command `bin2sgl -f build_arm/openrtx_dm1801_bin -m DM-1801` to encrypt it;
- you'll get a file named "openrtx_dm1801_bin.sgl" in the "build_arm" folder;
- flash it by issuing the command`./scripts/gd-77_firmware_loader.py -f build_arm/openrtx_dm1801_bin.sgl` from the OpenRTX root folder.

You should be able to flash the "openrtx_dm1801_bin.sgl" file using the OEM firmware update tool too.

silseva, 2022-06-17:
@jg1uaa commit 8accc3b should have fixed both the issues, can you please check and confirm?

Thanks!

jg1uaa, 2022-06-19:
Hi, sorry for late reply. Your procedure could produce openrtx_dm1801_bin.sgl without any errors (bin2sgl is compiled from open-ham repo).

Produced binary is not tested yet, I am looking forward to try it on my DM-1801.

silseva, 2022-11-11:
Closing this issue for inactivity, feel free to reopen it if necessary!

closed 2022-11-11 by silseva