OpenRTX/OpenRTX #101: 1750 Hz Tone

myzinsky, 2022-08-02:
Is there a possibility to generate a 1750 Hz tone yet? Otherwise I will look if I could contribute that :-)

For the background: German FM repeaters use this tone to open a repeater, instead of CTCSS.

silseva, 2022-08-02:
Hi!
There is already a tone generator driver up and running for MD-380/UV380 devices, so I'd say yes :)
The only small issue is to route properly that signal to the RF stage via the HR_C5000, but is nothing different from what is being done for M17. If you want to contribute, we'll be happy!

antoniovazquezblanco, 2023-07-30:
This seems to have been fixed. Could this be closed? :)

silseva, 2023-07-30:
Will do as soon as I merge Marco's contribution in the master branch!

antoniovazquezblanco, 2023-07-30:
Oops. Thanks!

turnrye, 2023-08-05:
I think we should keep this issue open until the follow-up is implemented as well. Quoting https://github.com/OpenRTX/OpenRTX/pull/141:
> This is currently only implemented for the MD-UV380, once the code is reviewed and OK, I can add the relevant logic to other radios.

So it's my impression that this is partially implemented in master, but needing support in the other builds and documentation updates. Please let me know if I'm wrong.

silseva, 2023-09-22:
Support for 1750Hz squelch tone is now in mainline and works for MD-3x0, MD-UV3x0, GD77 and DM-1801.
To transmit the tone, press the hash button while holding the PTT.

PR #141 (2023-07-19): Add 1750 Hz tone
When pressing the Hash key and the PTT key in FM mode a 1750 Hz tone is transmitted

This is currently only implemented for the MD-UV380, once the code is reviewed and OK, I can add the relevant logic to other radios.

This will resolve #101
closed 2023-09-22 by silseva