OpenRTX/OpenRTX #112: UV380G can tx but not rx on 477.125MHz

vk7js, 2022-09-08:
When testing OpenRTX on my UV380 with GPS, It seems that frequencies around 476-477 MHz, the Australian CB frequency range,  (within the 400-480 MHZ range the radio is supposed to support) the radio will tx but not rx on fm mode with and without CTCSS tones set.

silseva, 2023-04-21:
I'm going to investigate this soon by sniffing what the OEM firmware sends to the radio chip

blazeb5, 2023-10-24:
Hello,  I'm in the 438.800.000mhz on my hotspot and radios. I've moded an rt3s and uv380. The rt3s works fine but my uv 380 will not rx from the other radio or hotspot. I'm running the nightly build from 23oct2023 and same results with v.3.6 stable build of OPEN RTX...any help/advice?