OpenRTX/OpenRTX #113: pop when voice prompt starts playing

vk7js, 2022-09-15:
Hi Silvano,
 
I’ve drastically improved the smoothness of the voice prompts over the last couple of days but I have one remaining annoying pop I’d really love to fix when enabling the amp whenever a voice prompt is spoken.
I looked in the platform drivers/audio/ audio_MDx.c file and tried increasing the sleep in audio_enableAmp from 10 to 50 ms without seeing any improvement.
Obviously we don’t want to do that anyway, but I was just seeing if I could get an improvement by increasing the sleep since the comment there says it is to eliminate a pop. Do you have any ideas?
Perhaps increasing the volume of the amp slowly?
Thanks
Joe

silseva, 2022-09-17:
GPIO mapping for audio amp. enable and speaker mute seems to be correct, I have to investigate a bit deeper by looking at the signals with a 'scope.

vk7js, 2022-09-17:
Perhaps  we need to increase the volume over a few ms?

Joe

vk7js, 2022-09-18:
Hi Silvano,

I think silencing fm during voice prompt playback is probably higher priority than the pop.

I just saw Roger has released an alpha of OpenUV380 which is a port of OpenGD77 for the UV380. I wish his team wasn’t so closed. We all could have worked together to speed development up.

They support DMR already also, using the dmr code from the original firmware.

Let me know if there is something else I can help with to move OpenRTX along.

Joe

silseva, 2022-10-19:
Issue fixed during voiceprompts development

closed 2022-10-19 by silseva

vk7js, 2022-10-19:
I’ll rework it if you haven’t figured out what you broke, but I’ll need to do it tomorrow.

Joe

vk7js, 2022-10-19:
The bug in your latest head is that if vpStartTime is non-zero but less than getTickCount+50, you don’t return early.

This means that it doesn’t really accomplish anything which is why you get the crack.

Joe