OpenRTX/OpenRTX #119: Radio restarts after consulting info menu

7AAZ, 2022-11-16:
Upgraded RT3=MD380-VHF to the openrtx_md3x0_v0.3.5 version.
 In the menu, when I touch the info option, the radio reboots... . 
73s

CU7 AAZ

silseva, 2022-11-18:
Hi!
I tested the firmware on these radios, without issues:

- MD-380 UHF
- RT3 UHF
- MD-390 VHF

Unfortunately I don't have your exact radio model. In the next days I'll prepare a firmware image printing the radio info on the serial port, to that I can have more clues for debugging.

silseva, 2022-11-18:
Another question: the firmware image you have on the radio now is taken from the v0.3.5 release page (https://github.com/OpenRTX/OpenRTX/releases/tag/v0.3.5)?

7AAZ, 2022-11-18:
> Another question: the firmware image you have on the radio now is taken from the v0.3.5 release page (https://github.com/OpenRTX/OpenRTX/releases/tag/v0.3.5)?

Yes version 0.3.5

7AAZ, 2022-11-18:
> Hi! I tested the firmware on these radios, without issues:
> 
> * MD-380 UHF
> * RT3 UHF
> * MD-390 VHF
> 
> Unfortunately I don't have your exact radio model. In the next days I'll prepare a firmware image printing the radio info on the serial port, to that I can have more clues for debugging.

With version https://github.com/OpenRTX/OpenRTX/releases/tag/v0.3.4 , On my radio this issue does not happen
Big 73s

silseva, 2023-02-08:
Same question I asked for #120: what tool did you use to flash the firmware image?

silseva, 2023-02-10:
The bug seems to be related to the fact that the radio is flashed using the TYT firmware update tool, see #125

silseva, 2023-02-16:
Bug reproduced on an MD-380 after flashing the firmware with the TYT upgrade tool. I'm closing this and publish the next updates on #125.

closed 2023-02-16 by silseva