OpenRTX/OpenRTX #120: Possible problem on Accessibility menu

7AAZ, 2022-11-17:
Accessbility menu > Voice > OFF works well, Beep works well, when changed to level 1 rádio restart..
73s

silseva, 2022-11-18:
Hi!
I tested the firmware on these radios, without issues:
- MD-380 UHF
- RT3 UHF
- MD-390 VHF

Unfortunately I don't have the exact same model of your radio...

You can try restore the settings to their default values and see what happens. Also, if you can please send a picture of the label in the battery slot, I can gather a bit more infos for the debugging.

Thanks!

silseva, 2022-11-18:
Another question: the firmware image you have on the radio now is taken from the v0.3.5 release page (https://github.com/OpenRTX/OpenRTX/releases/tag/v0.3.5)?

7AAZ, 2022-11-18:
> Hi! I tested the firmware on these radios, without issues:
>
> * MD-380 UHF
> * RT3 UHF
> * MD-390 VHF
>
> Unfortunately I don't have the exact same model of your radio...
>
> You can try restore the settings to [...]

Hi,
Here goes the radio tag.
If you need anything else, I'm 100% available.

Hug !

[image]

silseva, 2023-02-08:
Back on this after a bunch of time. What tool did you use to flash the firmware image into the radio? The TYT firmware update tool, `radio_tool` or the dmr.tools website?

7AAZ, 2023-02-08:
Welcome back :). I used the TYT firmware update tool, like other times before.
Available for whatever you need
Thaks
73s

7AAZ, 2023-02-08:
> Back on this after a bunch of time. What tool did you use to flash the firmware image into the radio? The TYT firmware update tool, `radio_tool` or the dmr.tools website?
/
Welcome back :). I used the TYT firmware update tool, like other times before.
Available for whatever you need
Thaks
73s

silseva, 2023-02-10:
The bug seems to be related to the fact that the radio is flashed using the TYT firmware update tool.See https://github.com/OpenRTX/OpenRTX/issues/125

silseva, 2023-02-16:
Bug reproduced on an MD-380 after flashing the firmware with the TYT upgrade tool. I'm closing this and publish the next updates on #125.

closed 2023-02-16 by silseva