OpenRTX/OpenRTX #121: STM32F415VG drop-in replacement for hardware AES

philipag, 2022-11-24:
Hello all!  I am new to both OpenRTX/M17 and end user radios in general but have expertise in firmware/hardware development.

Is the STM32F405VG fast enough to also perform software AES CTR or similar mode so that the M17 data/voice encryption spec could be implemented in OpenRTX?  Or does this CPU not have sufficient additional bandwidth?  I was thinking that this would be possible since the encoded audio has a fairly low bit rate.  Also, I assume that ST provides mbedtls as part of their SDK?  So the real work would be implementing M17 encryption (the M17 spec talks about it but off-hand I did not find any sources that implement this).

If this is not possible, I believe the STM32F415VG would be a drop-in replacement with hardware AES support.

Just curious

silseva, 2022-11-25:
Hi!
I'm quite sure that the STM32F405 has enough grunt to support the AES encryption in M17, just we didn't implement it (yet). The main reason for the absence of AES is that we decided to bring up a working implementation of the protocol and leave the various additional features for the future development since, currently, the firmware is still missing some important parts, like the CPS system.
This said, if you want to bring up the AES part and make a pull request, your contribution is very welcome!

philipag, 2022-11-26:
@silseva Thanks for the info. I find this project really enticing but won't have any meaningful time to dedicate to it at the moment.  Waiting for a uv390 to arrive and will have a play with it though.

closed 2022-11-26 by philipag