OpenRTX/OpenRTX #123: [mod17-ui] CAN can't be changed in the menu

sp5wwp, 2022-12-09:
After entering the `Settings->M17 Settings` menu, CAN can't be changed nor set. Pressing the upper right button doesn't have any effect on the CAN menu entry.

silseva, 2023-02-08:
Can you provide a video of the problem, please? In this way I can try fix it even without having a Module17 myself.

sp5wwp, 2023-02-09:
OK, it can be set by using Left/Right buttons instead. Still, CAN and `Module17 Settings` appear to be volatile. They are not saved to flash.

silseva, 2023-02-09:
Yes, because the changes are still in the `mod17-ui` branch. I was waiting to fix (or for the fix) this last problem before merging everything into master.

silseva, 2023-02-13:
As from our discussion, we leave the value changeable with left/right buttons for consistency with the other settings entries.

closed 2023-02-13 by silseva