OpenRTX/OpenRTX #125: v0.3.5 crash when flashed using TYT firmware update tool

silseva, 2023-02-10:
After flashing the v0.3.5 binaries using the TYT firmware update tool the radio crashes on info menu and when changing the voice prompt levels to 1. The problem is not present if the binary is flashed either by using `radio_tool` or the dmr.tools website.

Crash when accessing info menu seems to suggest a truncated .data section during the flashing process.

silseva, 2023-02-10:
@7AAZ can you try flashing the v0.3.5 binaries either by using `radio_tool` or the dmr.tools website?
Alternatively, can you send me the executable of the TYT flashing tool? I've lost mine.

7AAZ, 2023-02-10:
> @7AAZ can you try flashing the v0.3.5 binaries either by using `radio_tool` or the dmr.tools website and see if the problem persists, please? Alternatively, can you send me the executable of the TYT flashing tool? I've lost mine.

@silseva download link for tyt flhasing tool - https://drive.google.com/file/d/14wkkBPxmIjtwC-mWz3Xl3QjcRkxjA4mN/view?usp=sharing

I will try flash with dmr.tools. I'll give feedback later

silseva, 2023-02-16:
Confirmed that the radio crashes only if the firmware is flashed using the TYT firmware update tool.

silseva, 2023-02-16:
[image]
Compare of two dumps of the MCU flash memory: the first of a binary flashed using `radio_tool`, the one of a binary flashed using the TYT upgrade tool. In the latter we see that the last 350 bytes have been truncated: this is a piece of the `.data` section containing the data structure used by `malloc`, which explains the two crashes.

@v0l may it be some kind of issue with the `radio_tool` wrapping process?

silseva, 2023-07-24:
It seems that this issue is related to the version of the firmware download tool used. The "Firmware Download" program V3.04 works fine, it can be downloaded here: https://www.hamdigitaal.nl/download/firmware/tyt_firmware/FirmwareDownloadV3.04_EN.exe