OpenRTX/OpenRTX #126: FYI. New UV380 and MD9600 have non-genuine MCUs

rogerclarkmelbourne, 2023-02-19:
FYI.

The non-genuine MCU's have one less bit in the register(s) which set the ISR priority, and the bit which is missing is the LSB

The solution is to only use either even, or odd, numbers for the ISR priority, and this will then work with the genuine and fake MCU
Otherwise different ISRs may be on the same priority e.g. 0 and 1 map to the same priority in the fake CPU, so we had to make sure we do not use adacent numbers

There may be other problems with the fake MCU, but we have not found any

silseva, 2023-02-26:
Hi Roger!
Thanks for the warning, I'll go through the code to see if some priorities needs to be updated. Also, in case we find some other bugs we'll keep you updated!

rogerclarkmelbourne, 2023-02-26:
No problem

I now changed my code so that I only use priority 0 to 7 and also changed number number of priority bits from 4 to 3

I'm not 100% sure of which bits in the config register contain the priority, I think possibly in the genuine chip its 

76543210
XXXX

But in the fake chip its only

76543210
XXX

silseva, 2023-03-17:
Interesting! We should decap one of those fake chips to see why they did so... :D

closed 2023-04-14 by silseva