OpenRTX/OpenRTX #127: Baofeng MD1702 branch ??

rogerclarkmelbourne, 2023-03-01:
Is this branch named correctly ?

https://github.com/OpenRTX/OpenRTX/tree/md1702

The commit for this is

[Add Baofeng MD-1702/DM-X target](https://github.com/OpenRTX/OpenRTX/commit/5800849cd7be2f6b089a93e9140d344bc3bb9ee2)

but there is no radio called a MD-1702

There is a DM-1702

Also, RadioTool does not support a DM-1702

https://github.com/v0l/radio_tool

But it does support a DM-1802 and DM-1701

Are these all the same radio?

silseva, 2023-03-03:
Is just a typo, it should be "DM-1702".

Thanks,
Silvano.

rogerclarkmelbourne, 2023-03-03:
OK.

Is there a way to load firmware onto the 1701?  RadioTool does not list it

https://github.com/v0l/radio_tool

BTW. If you want the pin map for the Baofeng DM-1701 I should know it soon.  It seems almost identical to the UV380 

I just ordered a DM-1701 to test it.

closed 2023-04-14 by silseva