OpenRTX/OpenRTX #128: Destination "ECHO" Does Not Echo

robrobinette, 2023-03-11:
When I transmit with my MD-UV380 with "ECHO" set as destination I get this in the live logs:
M: 2023-03-11 18:04:05.192 M17, received RF late entry voice transmission from K9OJ to ECHO
M: 2023-03-11 18:04:09.508 M17, **transmission lost** from K9OJ to ECHO, 4.3 seconds, BER: 1.8%

When I transmit using m17-mod-gui with "ECHO" set as destination I do get the echo reply and this in the logs:
M: 2023-03-11 18:06:20.280 M17, received RF late entry voice transmission from K9OJ to ECHO
M: 2023-03-11 18:06:24.326 M17, received RF end of transmission from K9OJ to ECHO, 4.1 seconds, BER: 0.0%
M: 2023-03-11 18:06:26.372 M17, received network voice transmission from K9OJ to ECHO
M: 2023-03-11 18:06:30.450 M17, received network end of transmission from K9OJ to ECHO, 4.1 seconds

When I transmit an echo test with my MD-UV380 I do not get an echo but when I follow up immediately with an echo test using m17-mod-gui I get repeated back to me both echoes, the MD-UV380 transmission and then immediately afterward the m17-mod-gui transmission.

silseva, 2023-03-12:
Hi!
Now we're always transmitting an EOT frame when PTT is released. I attached a new binary with the fix here, you can test if the ECHO function now works without having to wait for the next nightly build.
[firmware.zip](https://github.com/OpenRTX/OpenRTX/files/10951289/firmware.zip)

robrobinette, 2023-03-12:
I loaded the firmware and now ECHO destination works as it should. Here's the MMDVM live log during the ECHO transmission:
M: 2023-03-12 16:14:43.604 M17, received RF late entry voice transmission from K9OJ to ECHO
M: 2023-03-12 16:14:47.685 M17, received RF end of transmission from K9OJ to ECHO, 4.1 seconds, BER: 1.1%
M: 2023-03-12 16:14:49.731 M17, received network voice transmission from K9OJ to ECHO
M: 2023-03-12 16:14:53.848 M17, received network end of transmission from K9OJ to ECHO, 4.2 seconds

silseva, 2023-03-12:
Perfect!  We can consider the issue closed, then!

closed 2023-03-12 by silseva