OpenRTX/OpenRTX #129: Missing header causing "received RF late entry voice transmission"

robrobinette, 2023-03-12:
When I transmit to ECHO from my OpenRTX (3/11/23 nightly) MD-UV380G to my hotspot I get the following error:

M: 2023-03-12 16:14:43.604 M17, received RF **late entry** voice transmission from K9OJ to ECHO
M: 2023-03-12 16:14:47.685 M17, received RF end of transmission from K9OJ to ECHO, 4.1 seconds, BER: 1.1%
M: 2023-03-12 16:14:49.731 M17, received network voice transmission from K9OJ to ECHO
M: 2023-03-12 16:14:53.848 M17, received network end of transmission from K9OJ to ECHO, 4.2 seconds

It happens with every transmission. When I transmit M17 from a FTM-200 using the same hotspot:
M: 2023-03-12 16:22:32.032 M17, received RF voice transmission from K9OJ to ECHO
M: 2023-03-12 16:22:36.158 M17, received RF end of transmission from K9OJ to ECHO, 4.1 seconds, BER: 0.0%
M: 2023-03-12 16:22:38.207 M17, received network voice transmission from K9OJ to ECHO
M: 2023-03-12 16:22:42.326 M17, received network end of transmission from K9OJ to ECHO, 4.2 seconds 

The MD-UV380G header frame seems to be missed causing the "late entry" error message.

robrobinette, 2023-03-12:
I do see that my OpenRTX modded MD-UV380G initially transmits 25kHz low for about 0.2 second before it stabilizes on frequency which could cause the loss of the header. If other MD-UV380 users could use an SDR and zoom in on the waterfall to see if it's just my radio or not that would help.

silseva, 2023-03-17:
Is something related to the RF section, which already showed to have some quirks. I'm going to do some tests and then see how to fix this, probably inserting a small delay before starting the M17 transmission to allow the carrier to stabilize.

silseva, 2023-07-14:
While waiting for a proper solution, I implemented a workaround which fixes the problem. I effectively saw a carrier-only gap between the starting of the preamble and the subsequent M17 stream: putting a small delay between the HR_C6000 TX enable and the activation of the TX on the AT1846S is enough to make the gap disappear.

silseva, 2023-07-29:
@robrobinette did you have the chance to see if now the hotspot is still showing a late RF entry? From the tests I did with the spectrum analyzer and with another radio, the issue should have been fixed.

robrobinette, 2023-07-30:
I downloaded the nightly, flashed it and I get the same error for late entry on my repeater:

M: 2023-07-30 15:06:53.487 M17, received RF late entry voice transmission from K9OJ to ECHO
M: 2023-07-30 15:06:54.690 M17, transmission lost from K9OJ to ECHO, 1.2 seconds, BER: 1.2%, RSSI: -141/-141/-141 dBm

And the same from my hotspot:

M: 2023-07-30 15:09:16.707 M17, received RF late entry voice transmission from K9OJ to ECHO
M: 2023-07-30 15:09:18.184 M17, transmission lost from K9OJ to ECHO, 1.5 seconds, BER: 1.8%

silseva, 2023-07-30:
Thanks for testing!
As for the other issue (#144), I have to set up an MMDVM hotspot for further investigation.

robrobinette, 2023-07-30:
My UV380 that I flashed today shows in the "Info" page I'm running V0.3.4.9-q5a....  Is this correct for a fresh nightly? I thought 3.5.0 was the latest release? Maybe my attempts to flash nightlies has failed?

silseva, 2023-07-31:
Uh, no. It seems that the server from where you downloaded the nightlies is stuck. I downloaded and flashed an MD380 image from [our nightly server](https://files.openrtx.org/nightly/) and the version is `v0.3.5-57-g61c12a32`.
I'll try later on with an UV380 too, but it should work. We fixed the build script a couple of weeks ago.

robrobinette, 2023-07-31:
Sorry about using an old nightly. I flashed the nightly from your link (and deleted my old browser favorite) and verified the version number was 3.5.77. . .

I still get the "late entry" error but the "transmission lost" error has been eliminated.  Echo now works so I assume it was the "transmission lost" error that was preventing echo from working for me. Here is my live log from my repeater doing an echo transmission:

M: 2023-07-31 14:32:00.254 M17, received RF late entry voice transmission from K9OJ to ECHO
M: 2023-07-31 14:32:04.419 M17, received RF end of transmission from K9OJ to ECHO, 4.2 seconds, BER: 0.1%, RSSI: -141/-141/-141 dBm
M: 2023-07-31 14:32:06.469 M17, received network voice transmission from K9OJ to ECHO
M: 2023-07-31 14:32:10.666 M17, received network end of transmission from K9OJ to ECHO, 4.2 seconds

silseva, 2023-08-01:
No need to apologize! Do you remember from where you downloaded the "broken" nightlies?
I'll keep this issue open for the late entry problem and close #144 as solved.

robrobinette, 2023-08-01:
No, I deleted the browser link. It was a website with a ham callsign in it though.

commit a5b925e7 (2023-07-14): Small code cleanup in MD-UV3x0 HR_C6000 driver, temporary workaround to fix #129