OpenRTX/OpenRTX #132: Squelch level too low

F4FYF, 2023-04-11:
Hi

I've just install OpenRTX Firmware on a RT3s with GPS.. Everything is allright but there is just the squelch level who is too low and open more often in analogique mode. Is there a tips to control it ?
Thanks for all

Jeff - F4FYF

silseva, 2023-04-12:
Hi Jeff!
To adjust the squelch level you have to press the lowest side button to enter in the "macro" menu and then use either the channel selector knob or the arrows on the keypad to increase/decrease the level. Let me know if it worked!

Silvano IU2KWO

F4FYF, 2023-04-12:
Hi Silvano

It's perfect ! everything works correctly and the squelch method is good. Thank you very much for everything.

Jeff - F4FYF

silseva, 2023-04-12:
Nice, I'm happy for that! :D

73,
Silvano IU2KWO.

closed 2023-04-14 by silseva