OpenRTX/OpenRTX #133: RT3S issue

iz4tow, 2023-04-12:
Good morning,
first of all you did a great job and I hope you'll continue to develop this software.

However I have an issue with RT3S, when I reboot my device I lost all settings change. I tried to use both last release binary and to compile the last version from sourcecode but no difference.

Thanks a lot

silseva, 2023-04-12:
Hi!
Can you tell me which firmware version are you using, please? You can find it at the "info" page of the menu, is the first line displayed.

Thanks!

iz4tow, 2023-04-13:
V0.3.5-3-g4a3bc93 and V0.3.5

silseva, 2023-04-15:
Sorry I forgot to ask another thing: what settings do you lose? All the settings, or just some of them?
For example, the M17 destination is not yet being saved in the flash memory.

iz4tow, 2023-04-15:
The settings I tried to change:
- gps settings
- m17 callsign
- frequency
- uhf/vhf
I lost all of them

silseva, 2023-04-16:
This is really strange, can you please tell me the serial number of the radio? I'm suspecting it has some kind of different hardware from the usual ones.

Thanks!

iz4tow, 2023-04-16:
1) 2105R03S1104408
2) 2105R03S1104403

I don't know if it's important but to flash the firmware I'm using radio_tool

iz4tow, 2023-04-17:
Yes both. I'll try to check the MCU in the afternoon

silseva, 2023-04-18:
I tested both the binary from the v0.3.5 release and one compiled from the latest master and I didn't see the issue. I'm afraid the issue is related to the hardware of your radio.

TechMindsYT, 2023-07-21:
I also experience the same issue. Unfortunately the writing from the MCU is scrubbed off. I just received the RT3S today, so maybe its a later version.

antoniovazquezblanco, 2023-07-27:
I can confirm this is also happening to me with a Dec 22 RT3S...

silseva, 2023-07-28:
I need your help: please flash the firmware image attached below, go to the info page and tell me the values of the last three fields (IDCODE, OPTCR and OPTCR1). I suspect that the new RT3s have a GD32 MCU which, in turn, have a slightly different management of the flash memory, where we store the settings.

Thanks!

[openrtx_mduv3x0_test.zip](https://github.com/OpenRTX/OpenRTX/files/12193317/openrtx_mduv3x0_test.zip)

antoniovazquezblanco, 2023-07-28:
Will do this afternoon!

antoniovazquezblanco, 2023-07-28:
[image]

Oopsie...

silseva, 2023-07-28:
Sorry, I prepared the binary while at work and in a rush. The one attached here should word, I tested it on an MD-380.
Btw, the ID code of my radio, with an STM32F405 is 0x10076413.

[openrtx_mduv3x0_test.zip](https://github.com/OpenRTX/OpenRTX/files/12196851/openrtx_mduv3x0_test.zip)

antoniovazquezblanco, 2023-07-28:
Well, the data above was right:
IDCODE 0x0009a413
OPTCR 0x0fff55ed
OPTCR1 0x00000000

That IDCODE indicates a Signia Technologies ARM core but that is rare...

silseva, 2023-07-28:
Definitely is not a true STM32 device, also with respect to what is written [here](https://stm32f4-discovery.net/2014/12/library-44-stm32f4-unique-id-and-flash-size/). I'm getting more and more convinced that we're dealing with a GD32 chip, I'll look at the reference manual and see if there are some differences with respect to the STM32 regarding the flash management

antoniovazquezblanco, 2023-07-28:
I am not familiar with how you read the idcode. Would you mind pushing the code to the repo? It may be a nice feature to even have in the master branch... I want to read it if possible. I will try to look for information on that ID but it is really strange. If that ID follows ARM speciation it is not from GD...

I am away from my tools but when I get back I will try to tinker with the SWD to attempt to read the ID via JTAG...

Please, let me know if I can do anything else to help locating more info.

Thanks!

antoniovazquezblanco, 2023-07-28:
In case that helps, the package of the component seems to be LQFP100 if I am not mistaken :)

antoniovazquezblanco, 2023-07-28:
I was unable to find an ID match in the M4 GD device manuals.

silseva, 2023-07-28:
A bit of time ago I read somewhere on the internet (unfortunately I don't remember where) that the GD32 MCUs have the same device ID of the STM32 ones (lower 11 bits of the register) but a "random" revision ID (bits 23 - 16). The device ID is indeed the same, 0x413, while the rest is different.
About how I read the IDCODE, nothing more than a `snprintf(buf, max_len, "%08x", DBGMCU->IDCODE);`. In any case the register is located at address 0xE004 2000, which is in the range of the ARM architecture registers.

PS: when you have time, can you please test the firmware attached below? I did a couple of small changes in the code which writes into the flash memory, let's see if now it works.

[openrtx_mduv3x0.zip](https://github.com/OpenRTX/OpenRTX/files/12197870/openrtx_mduv3x0.zip)

antoniovazquezblanco, 2023-07-28:
New settings info:

[image]

Settings are still not persistent though....

TechMindsYT, 2023-07-28:
Here's my  RT3s, just uploaded the firmware above and took this screen shot if it help?

[image]

silseva, 2023-07-29:
Thanks for you help, guys! 
Now we know that both of your radio have the same MCU and that MCU is not an STM32.

I have to ask you to do another test, please: the firmware attached below tries to erase the flash sector 11 and write some random data to it. I have to see what goes wrong in the erase/write process.

[flash_test_mduv3x0.zip](https://github.com/OpenRTX/OpenRTX/files/12206619/flash_test_mduv3x0.zip)

antoniovazquezblanco, 2023-07-29:
[image]

TechMindsYT, 2023-07-29:
Here's mine..

[image]

silseva, 2023-07-29:
Another round of testing, thanks two my friends we probably got the point!

[flash_test_v2_mduv3x0.zip](https://github.com/OpenRTX/OpenRTX/files/12207278/flash_test_v2_mduv3x0.zip)

TechMindsYT, 2023-07-29:
Here you go

[image]

silseva, 2023-07-29:
Fixed!! I'm going to push the commit asap.
Thanks for the help, guys!

TechMindsYT, 2023-07-29:
Nice! could you share the fixed firmware here? pls.

silseva, 2023-07-29:
Here it is!

[openrtx_mduv3x0.zip](https://github.com/OpenRTX/OpenRTX/files/12207371/openrtx_mduv3x0.zip)

TechMindsYT, 2023-07-29:
Thank you for your work, I can confirm that my RT3S is now saving settings! You the man! Cheers!

silseva, 2023-07-29:
Thanks for your help!

antoniovazquezblanco, 2023-07-29:
Awesome. Will test but I'm sure it is all going. Thanks for the sprint. Top notch support :)

SuperGlobal, 2023-09-29:
I am still having strange interaction when it comes to settings being saved on my radio.

Going from the stock RT3S firmware directly to the nightly OpenRTX firmware would not retain any settings (callsign, GPS settings, voice prompt) nor any of the entered channel information (macro menu, TX/RX frequency, protocol, squelch) on power cycle.

Once I stumbled upon this thread I downgraded from the nightly build to the final patched firmware that @silseva posted. This still would not allow me to save any of my settings (callsign, GPS, voice prompt, squelch, etc).

I finally uploaded the IDCODE debug firmware included from this thread to compare to the other users. I saw I had the same IDCODE as the other two people in this thread. I gave the final patched firmware another test and saving settings/macro menu was working! The only settings that I saw was not retained was the entered frequencies/M17 reflector name.

I tested a bunch of variations of upload paths to see what the differences were and included them below:

STOCK RT3S FIRMWARE -> NIGHTLY FIRMWARE
- Settings NOT retained
- Macro settings NOT retained
- Entered freqency/reflector NOT retained

NIGHTLY FIRMWARE -> PATCHED FIRMWARE
- Settings NOT retained
- Macro settings NOT retained
- Entered freqency/reflector NOT retained

IDCODE DEBUG FIRMWARE -> PATCHED FIRMWARE
- Settings retained properly (callsign, GPS, voice prompt)
- Macro settings retained properly (protocol, squelch)
- Entered frequency/reflector NOT retained

IDCODE DEBUG FIRMWARE -> PATCHED FIRMWARE -> NIGHTLY FIRMWARE
- Settings NOT retained
- Macro settings retained
- Entered frequency/reflector NOT retained

IDCODE DEBUG FIRMWARE -> NIGHTLY FIRMWARE
- Settings retained properly
- Macro settings NOT retained
- Entered frequency/reflector NOT retained

silseva, 2023-09-30:
@SuperGlobal can you try the firmware attached here, please, and report the IDCODE of the MCU? https://github.com/OpenRTX/OpenRTX/issues/133#issuecomment-1656084919

SuperGlobal, 2023-09-30:
For sure, I've attached the image and attached it to this reply!

It is the same as what's mentioned in this [comment](https://github.com/OpenRTX/OpenRTX/issues/133#issuecomment-1656156689).

[image]

SuperGlobal, 2023-10-01:
Okay there are a couple things going on here. The initial nightly firmware that I upgraded my stock RT3S to was the firmware hosted by DF5PMF which from reading on the Matrix is still building OpenRTX from a commit back in September 2022 and when I wrote my comment the nightly firmware I was upgrading to was the firmware hosted on openrtx.org which was my bad I thought they were both the same!

After realizing this I was able to drill down a couple things:

- The macro menu AND settings that are set on first boot are retained/persistent but any other modifications to the macro menu AND settings after first boot will result in settings being reverted to those set first boot.
- The entered frequency acts the same exact way as the macro menu/settings. The frequency entered at first boot is retained/persistent but if adjusted after first boot the frequency will be reverted to the frequency set first boot.
- For M17 mode I was never able to get a different reflector name to be retained/persist across power cycles.

Steps to demonstrate settings being reverted:
1. Write OpenRTX firmware to RT3S
2. On first boot of fresh firmware install, configure macro menu/squelch/mode/callsign/settings/frequency
3. Power cycle radio
4. Change any of the previously configured options such as frequency of squelch
5. Power cycle radio
6. Confirm previously configured options were reverted to those set first boot

silseva, 2023-10-02:
I suspect that the MCU in your radio wants the flash to be fully erased before each write. I'll prepare a test firmware later on to confirm this behavior.

silseva, 2023-10-08:
I created a new issue dedicated to the last firmware problems on the new RT3s devices, discussion follows up on #198.

closed 2023-10-08 by silseva