OpenRTX/OpenRTX #134: RT3S bug on TX/RX

iz4tow, 2023-04-14:
Hi
I tried both V0.3.5-3-g4a3bc93 (compiled from source) and V0.3.5 (downloaded from release link), I tried to transmit/receive with 2 RT3S in UHF and VHF but I can't receive nothing from the listening radio.
Using RTL SDR I can see the signal, so I think the problem is on RX side.
Can you help me?

silseva, 2023-04-14:
I'll do some tests in the next days. If you are using analog FM a possible cause can be the CTCSS squelch in RX: to disable it make sure that the item number 3 in the quick menu (the one you open by pressing the side button) is either empty or set to "E".

iz4tow, 2023-04-14:
Sorry, I didn't say that I'm trying to use m17. To activate it I click on macro button(quick menu?), then 5 on both rt3s then I use the PTT button to talk.
The issue is only on m17, analog FM  works

silseva, 2023-04-15:
In this case a possible cause is the mod wire going from the RF side of the PCB to the microcontroller. If it gets pinched between the PCB and the metal case it is shorted to ground and the MCU does not receive any baseband signal.
Check if, when you transmit M17, on the receiving radio the green led turns on: if so, a valid signal is being received.

iz4tow, 2023-04-15:
Yes on the Rx side the green led is on during the transmission and also the receiving volume bar on the bottom (the white bar)
I'm sure that tx side is working because on sdrpp I can see the modulated m17 signal.

silseva, 2023-04-17:
I'm going to run a couple of tests tomorrow to see if I can reproduce the issue.

silseva, 2023-04-18:
I confirm there are some issues with M17 RX, also on MD-380

silseva, 2023-08-30:
From the tests I did the audio issue should have been fixed with the latest audio driver rework. Working firmware will be available starting from tomorrow's nightly builds.

closed 2023-08-30 by silseva

iz4tow, 2023-11-27:
Still not working

iz4tow, 2023-12-04:
I wrote you a week ago but I got no response

silseva, 2023-12-06:
Interesting, I got nothing. To which address did you write?
My address is silseva@fastwebnet.it