OpenRTX/OpenRTX #142: About latching of the macro menu

marcoSchr, 2023-07-02:
According to the source code and my observations the latching of the macro menu should behave as follows:
- When pressing the macro button long, the menu latches
- Settings in the macro menu can be changed, buttons that do not change the macro menu are ignored
- 3 seconds after the last button press, the menu can be unlatched by any button press, and the button press will be interpreted as if not in the macro menu

I think this is not very intuitive. As a user I would expect a few things:
- When the macro menu is latched, at any time I can close it again with the macro button
- When the macro button is latched, the button press to unlatch the macro menu should not be processed

I propose the following changes to make the macro menu more intuitive:
- Pressing the macro button and quickly selecting a setting should behave as before, changing the setting and closing the macro menu again
- Pressing and holding the macro button should lock the macro menu
- Any settings can now be changed, any button not related to the macro button should be ignored
- Pressing the macro button again should unlock and hide the macro menu again

Additionally I noticed, that when I turn the radio on and latch/unlatch the macro menu multiple times, after a few times it will no longer latch, but it will still beep.

silseva, 2023-07-30:
Hi!
The macro menu latching was implemented by VK7JS as an help for the vision impaired operators. This said, I concur that the actual implementation is not super-intuitive, your proposal is good and I'd be happy if you can contribute with its implementation. I'd just add a timeout mechanism to unlatch and hide the macro menu after, say, three seconds of inactivity.

silseva, 2023-08-17:
Improved behavior of the macro menu brought in with #156.

PR #156 (2023-08-17): Macro latching
This merge request will change the behavior of the macro latching

Now when the macro button is held it will latch the macro menu.
The letter L will be displayed in the upper left corner to show the latched status.
When the macro menu is latched, the macro button will unlatch the menu.

It would be good if some one who is vision impaired can check, if the menu is still usable with this change.

Fix #142
closed 2023-08-17 by silseva