OpenRTX/OpenRTX #143: Watt Level Changes With Mode Change

robrobinette, 2023-07-09:
I cannot set 1 watt on both M17 and FM modes. When I set 1 watt on either mode then change to the other mode that mode is at 5 watts. If I change it to 1 watt then when I go back to the other mode is at 5 watts.

I also cannot set 5 watts on both modes. One is always 1 watt and the other is at 5 watts.

I am using an OpenRTX daily from about 3 July 2023.

Rob Robinette K9OJ

robrobinette, 2023-07-09:
This happens with my OpenRTX MD-UV380.

silseva, 2023-07-14:
I did some tests using today's nightly build on an MD-380 (single band) and I didn't see the issue. I'll try later on with an RT3s to see if there are differences.
Also, the nightlies were stuck to some commit in the past for a while and we fixed them the 8th of July: please try again downloading a firmware image and see if the issue is still present. I cannot exclude we already fixed it but you didn't have the fix due to the stuck nightlies.

73,
Silvano IU2KWO.

robrobinette, 2023-07-18:
I loaded a new nightly on 15 July and the bug is still there.

silseva, 2023-07-29:
Did you refer to the actual output power or to the value displayed in the macro menu?
In the second case, can you please send a small video of it?

robrobinette, 2023-07-30:
I went by the macro menu, I did not test the actual radio output.
I will flash the latest nightly and try it. If it still happens I will do a video.

robrobinette, 2023-07-30:
Same behavior with today's nightly. If I change the output power in one mode, the other mode switches too. One will have 1W and the other mode will have 5W. Change the 5W to 1W and the other mode will switch to 5W. I'm trying to get my iPhone video into a small format to allow upload.