OpenRTX/OpenRTX #144: EOT Frames Missing

robrobinette, 2023-07-09:
My modified MD-UV380 always shows a "transmission lost" error message in the MMDVMHost log file when transmitting M17 voice. Because of this "echo" doesn't work for the UV380. If I follow up with a transmission from my digirig/FTM-200DR to echo then echo will send both the UV380 and the FTM-200DR echo messages.

MD-UV380 MMDVMHost log during echo transmission:
M: 2023-07-06 11:13:20.934 M17, received RF late entry voice transmission from K9OJ to ECHO
M: 2023-07-06 11:13:28.616 M17, **transmission lost from K9OJ to ECHO, 7.7 seconds, BER: 0.4%, RSSI: -141/-141/-141 dBm**

digirig/FTM-200DR MMDVMHost log echo transmission immediately following the UV380:
M: 2023-07-06 11:16:42.068 M17, received RF voice transmission from K9OJ to ECHO <from FTM-200DR>
M: 2023-07-06 11:16:44.831 M17, received RF end of transmission from K9OJ to ECHO, 2.8 seconds, BER: 0.3%, RSSI: -141/-141/-141 dBm <from FTM-200DR>
M: 2023-07-06 11:16:46.877 M17, received network voice transmission from K9OJ to ECHO <echo from UV380>
M: 2023-07-06 11:16:56.026 M17, network watchdog has expired, 7.7 seconds
M: 2023-07-06 11:16:56.043 M17, received network voice transmission from K9OJ to ECHO <echo from FTM-200DR>
M: 2023-07-06 11:16:57.320 M17, received network end of transmission from K9OJ to ECHO, 1.3 seconds

The EOT frames from the UV380 seem to be missing.

I'm running an OpenRTX daily from about 3 July 2023.

Rob Robinette K9OJ

silseva, 2023-07-14:
I'm going to set up an MMDVM hotspot as soon as possible to debug this issue properly.

Silvano IU2KWO

silseva, 2023-08-01:
From the discussion in #129 we can consider this issue solved. I'll wait for confirmation before closing.

robrobinette, 2023-08-01:
I agree, I will close the issue.