OpenRTX/OpenRTX #151: M17 volume control doesn't work on TYT MD380 UHF

turnrye, 2023-08-05:
Hardware: TYT MD380 UHF
Build: e689e2f7ae53f15c2828bc5edeb8ed801ddd8050

## Steps To Reproduce

1. Tune to an M17 signal
2. Turn the volume knob up and down

## The current behavior

Rotating the volume knob to increase it does not increase the volume of the decoded voice, but instead just a background noise is increased

## The expected behavior

Rotating the volume knob should increase the volume of the decoded voice, just as it behaves when tuned to analog FM

silseva, 2023-08-22:
Is a general issue with M17 due to how the demodulated audio is being reproduced. A fix is in progress.

silseva, 2024-10-12:
Issue fixed on MD-UV380 by using the HR_C6000 DAC for audio playback

closed 2025-09-05 by turnrye