OpenRTX/OpenRTX #155: Switch font to more aesthetically pleasing option

heylix, 2023-08-08:
Currently we're using Ubuntu and FreeSans, which are not exactly optimised for pixel display.

What do you think about switching this to a more aesthetically pleasing pixel/bitmap font?

For example, we could use [Gohu Font 11px](https://font.gohu.org/) which has a more sleek "M" in my opinion.

Another idea would be [Proggy](https://github.com/bluescan/proggyfonts/) which has nice rounded characters.

I think with a non-monospace bitmap/pixel font we can fit quite some information on the screen, but in some cases a mono font would probably also be a good idea.

silseva, 2023-08-22:
Paging @n1zzo and @Nimayer for this

LolloDev5123, 2025-04-22:
I am on par with your opinion. I think a part of development focus should go to having very pleasing UX, especially where there is opportunity in color LCDs.

IMO, the Kenwood TH-D74 has stunning UI