OpenRTX/OpenRTX #157: Create GUI application to install OpenRTX on radios

Nimayer, 2023-08-09:
# Goal
To reduce the difficulty barrier for new users we need a GUI application to install OpenRTX.

## Requirements
* The application should run on Linux, Windows and macOS
* The application should be a standalone executable, and should not require installation
* The application should be able to perform these operations on all supported targets:
  * Download latest/other release of OpenRTX
  * Flash OpenRTX
  * Create EEPROM backup
  * Create calibration data backup
  * Format EEPROM with littleFS
  * Restore EEPROM from backup
  * Flash manufacturer firmware (provided by user)

## Choices
* Rust has been chosen as language, to match [rtxlink](https://github.com/OpenRTX/rtxlink)
* [CXX](https://cxx.rs/) is used as interface with [radio_tool](https://github.com/v0l/radio_tool), a cli radio flashing software

## Reference
* [gnome-firmware](https://gitlab.gnome.org/World/gnome-firmware) is a design and layout reference, for it's ability to show  several supported devices at once.

turnrye, 2023-08-12:
https://github.com/OpenRTX/rtxflash and https://github.com/OpenRTX/OpenRTX-Companion are the future homes for this effort.

silseva, 2023-08-17:
Given that the development is ongoing on the dedicated repositories, should we close this?

closed 2026-02-26 by turnrye