OpenRTX/OpenRTX #158: Support OpenRTX-defined Codeplug

turnrye, 2023-08-11:
# Motivation

Today, OpenRTX firmware reads the existing codeplug from the device. This means that the various hardware platforms have their own special codeplug handling. This is a good user experience, because users don't lose their codeplug when they adopt OpenRTX. Flashing OpenRTX is a single step, too. Unfortunately, it is incompatible with the vision of OpenRTX, which includes bringing novel capabilities to these hardware platforms, and these capabilities are not supported by the existing codeplug formats. For example, the first of such challenges is supporting the M17 mode. Its channels and contacts are substantially different from DMR, and as a result a new codeplug structure is needed.

## Requirements

- Define a new `.rtxc` filetype that represents this new codeplug format
- Support analog FM, DMR, and M17 modes
- Be thoroughly documented to ease adoption by other projects, like https://github.com/hmatuschek/qdmr
- Efficiently pack the data, but not so much that it presents a burden to developers
- When flashing this new codeplug format, make it easy to go back to the original codeplug using backups

## Choices

- Having a text-based codeplug representation is desirable, but not critical path for the initial implementation
- Implementing memory management functionality within the firmware is not part of the critical path for this task

## References

- https://wiki.izzo.sh/openrtx_cps# and https://github.com/OpenRTX/OpenRTX/blob/master/openrtx/include/core/cps.h are documentatino for an existing codeplug structure proposal
- https://github.com/hmatuschek/qdmr/issues/251 is an existing work-in-progress integration in a CPS related to this effort

antoniovazquezblanco, 2023-09-04:
This is really nice!

I have just found the `cps.h` (https://github.com/OpenRTX/OpenRTX/blob/master/openrtx/include/core/cps.h) file. I was wodering if it makes sense to transfer this file into a separate repo in order to provide an implementation that other programs may reuse to reduce implementation burden.

Thanks!

silseva, 2023-10-14:
Currently there is an RFC open for some changes to the OpenRTX codeplug specification. The new version, open to comments is available here: https://github.com/OpenRTX/openrtx.github.io/blob/cps_refactor/binary_cps_format.md

Comments and discussions to be added here: https://github.com/OpenRTX/openrtx.github.io/pull/32

Movin the codeplug specification and header files in a separate repository can be a good idea, we can think about doing so once we have the first stable version.

closed 2026-02-26 by turnrye