OpenRTX/OpenRTX #160: No audio until vol control turned to halfway and high minimum audio level MD-UV380 NO-GPS OpenRTX v0.3.5

edmunster827, 2023-08-13:
This doesn't happen on 0.3.4. From power on there is no audio unless the vol pot is set to or turned at least halfway on. Once audio is present it behaves normally, except for the high minimum audio level. On 0.3.4 the audio is ok from power on and the minimum is reasonably quiet. 
73 and thanks.

turnrye, 2023-08-15:
I've reproduced this behavior as well. Thanks for the bug report!

silseva, 2023-08-30:
I found the cause of the problem, I'll fix it as soon as I can.

edmunster827, 2023-08-30:
Great, glad to have helped. Unfortunately I won't be able to test, I killed the radio in a late night soldering accident trying to mod for m17. The instructions are great, but doing such fine work tired is not recommended. It is now deaf on all modes, I grounded the hr_c6000 inv_in pin. 🤦 Keep up the good work!

commit 5ea94ee1 (2023-09-08): Improved management of FM audio volume on MD-UV3x0.

Fix #160.
closed 2023-09-08 by silseva