OpenRTX/OpenRTX #171: FM rx broken after TX - recent RT3s. Bad commit found.

tarxvftech, 2023-08-20:
Recent RT3S/mduv3x0 with GPS, current master branch, in FM mode.
There's some kind of bug preventing the rf receiver from playing the correct audio sometimes. 
﻿
It seems to happen very reliably, every time I push PTT for example.
Once it happens it seems to require a reboot to clear. Adjusting squelch and opmode doesn't seem to make a difference in whether it happens or in clearing the issue. Once it happens, the speaker is playing a low hum/buzz when squelch is open, and is silent when squelch is closed. Even after this, beeps and voice prompts work perfectly. Volume works fine, and the beeps, voice prompts, and erroneous buzz/hum are adjusted in volume as you'd expect.
﻿
git bisect (amazing tool!) shows it's related to a5b925e76f98cbed2a019296d960d4a152e7a288 `Small code cleanup in MD-UV3x0 HR_C6000 driver, temporary workaround to fix #129`. 

I don't really know what I'm looking at with that diff. It does seem that with that commit from mid July, and every commit since, I can reproduce the issue readily. Before that commit, I cannot reproduce it at all.

tarxvftech, 2023-08-20:
I can confirm that current master with just that commit reverted fixes the bug.

turnrye, 2023-08-20:
I've also had an issue with no audio RX on my MD-UV3x0 with latest master (to be honest I've never had RX audio working right since I learned about this project a couple of weeks ago). My missing RX audio is always -- not just after TXing as encountered by @tarxvftech . Despite my reproduction steps being different, reverting a5b925e76f98cbed2a019296d960d4a152e7a288 resolves my issue as well.

tarxvftech, 2023-08-21:
In my case it's the HR_C6000_UV3x0.cpp M::CONFIG 0x26 register - removing the register write with value 0xFE in startAnalogTX, or setting it back to 0xFD in stopAnalogTX, either method fixes the issue for me.

tarxvftech, 2023-08-21:
```
diff --git a/platform/drivers/baseband/HR_C6000_UV3x0.cpp b/platform/drivers/baseband/HR_C6000_UV3x0.cpp
index a510cc4b..f6cc44ff 100644
--- a/platform/drivers/baseband/HR_C6000_UV3x0.cpp
+++ b/platform/drivers/baseband/HR_C6000_UV3x0.cpp
@@ -230,6 +230,7 @@ void HR_Cx000< M >::stopAnalogTx()
     writeReg(M::CONFIG, 0xE2, 0x06);    // Mic preamp disabled, anti-pop enabled
     writeReg(M::CONFIG, 0x34, 0x98);    // FM bpf enabled, 25kHz bandwidth
     writeReg(M::CONFIG, 0x37, 0x81);    // DAC gain
+    writeReg(M::CONFIG, 0x26, 0xFD);    // Undocumented register FE Turns off FM receive FD turns it on maybe?
 }
```

tarxvftech, 2023-08-22:
@silseva I understood the majority of our usage of this chip comes from some poor documentation and RE of the OEM firmware. I assume from that, that the setting of 0x26 to 0xFE on transmit has a purpose, and therefore we should keep it. 

Does then the above patch look reasonable to you?

silseva, 2023-08-22:
My fault, I changed the code in the driver without proper testing... 
The original firmware sets register 0x26 to 0xFE before transmitting and restores it to 0xFD at the end of transmission, your patch is good.

silseva, 2023-08-22:
[openrtx_mduv3x0.zip](https://github.com/OpenRTX/OpenRTX/files/12407367/openrtx_mduv3x0.zip)
Fixed firmware, if someone can test it, I'll then push the fix. I don't have an RT3s at hand, sorry.

tarxvftech, 2023-08-22:
Works fine!

commit 49209218 (2023-08-27): MD-UV3x0: fixed bug causing no FM rx after tx end. Fix #171

Thanks to turnrye and tarxvf for debugging and testing this.